sidoh/esp8266_milight_hub #524: More then 4 groups per remote

poudenes, 2019-09-19:
Hi,

I know that official remotes have maximum 4 groups.
Is there maybe a way to extend this? In Settings area where you can select max groups per virtual remote?

I don't use any official remotes, only HA with MQTT connection.
Reason of this question is:

I have 6 bulbs in bedroom, now I need 2 virtual remotes to control them
(In every group I haver 1 bulb, so I can control them individual in HA)
But when I use group_0 i have to do it for 2 groups

When you have for example 8 groups then 8 bulbs and 1 group_0

Same for living room (6 bulbs) and Diningroom (10 Bulbs)

sidoh, 2019-09-19:
The number of groups is unfortunately a limitation of the milight protocol.  The way that the packets are structured only allows for a certain number of groups.  The FUT089 protocol allows for 8 groups.

poudenes, 2019-09-19:
And your wonderful image does not support the FUT089?
Or is 8 groups max possible?

sidoh, 2019-09-19:
FUT089 is supported.  :)

sidoh, 2019-09-21:
@poudenes, gonna close this since it seems like your question is answered.  Let me know if I've misunderstood.

closed 2019-09-21 by sidoh