sidoh/esp8266_milight_hub #525: Question: unable to get MQTT messages on action with milight remote.

ReinierGielen, 2019-09-19:
When I enable sniffing I detect every action of the remote.
But actions on the remote do not result in MQTT messages in HASSIO.

In settings I have set:
update patern:  milight/updates/:device_id/:device_type/:group_id
state patern: milight/states/:device_id/:device_type/:group_id

I am not sure what I am doing wrong.
 I am expecting that the sniffed actions are forwardes as MQTT updates.
Is that correct ?

ReinierGielen, 2019-09-20:
I managed to get a response to the remote in HASSIO.
Here is my automation:
`- alias: MiLight Forwarder
  initial_state: True
  trigger:
    platform: mqtt
    topic: milight/states/0x8754/#
  action:
  - data:
      entity_id: light.eetkamer
    service: light.toggle  `

The id of my remote I sniffed being 0x8754
With this any button on the remote toggles the light.Eetkamer

So this is very nice. But also confusing because the trigger is states not updates.
[image]

But I it is working so I am not complaining.
Maybe this info is usefull for others having trouble controlling the remote

ReinierGielen, 2019-09-20:
`- alias: MiLight Forwarder
  initial_state: True
  trigger:
    platform: mqtt
    topic: milight/states/0x8754/+/1/#
  action:
    - service: mqtt.publish
      data_template:
        topic: "milight/0x3/rgb_cct/1"
        payload_template: >
          {{ trigger.payload }}`
This keeps my entity light.Eetkamer in perfect sync when remote is used.

I don't think I can write this generic because the id of the entity is not in the state message of the remote.

ReinierGielen, 2019-09-20:
Only 1 more thing to do.
Is it possible for the ESP to send a MQTT message when it loses or gains power ?
If so I coud update the availability of the lamps in HASSIO.

sidoh, 2019-09-20:
You can put the device ID, etc in the payload.  Check out the "group state fields" setting at the bottom of the MQTT settings pane.

I'd recommend turning off MQTT updates if you don't need them.

> Is it possible for the ESP to send a MQTT message when it loses or gains power ?

Yes, that's what the client status topic does.  See the [relevant section in the README](https://github.com/sidoh/esp8266_milight_hub#client-status) for further detail.

There's [quite a bit of information](https://github.com/sidoh/esp8266_milight_hub#mqtt) about the MQTT integration in the README.  I know it's a lot to wade through, but I expect everything you're looking for is there.

ReinierGielen, 2019-09-20:
You mean thuis part:
To receive updates when the MQTT client connects or disconnects from the broker, confugre the mqtt_client_status_topic

I tried that and it works. But it is not what ik want.
It sends when the ESP connects and disconnects from the broker.
What I want is to know from each individual light does it have power of not.

sidoh, 2019-09-20:
That's in state updates.  Perhaps what you're trying to do is covered here?

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

sidoh, 2019-09-20:
The payload will contain a status which you can switch on

sidoh, 2019-09-20:
oh, i may have totally misunderstood your question.  Are you asking if it's possible to know if external power to the bulbs is removed?  If so, no, that's not possible.  Communication with bulbs is entirely one way -- bulbs only send communication in some really weird cases (see #91) that aren't really helpful here.

ReinierGielen, 2019-09-21:
A solution may be to place a mains powered zigbee device in the circuit. And the check avaiability of that device.

sidoh, 2019-10-05:
(Sorry for the late reply)

That's a great idea!  Seems like the logic for that is best contained elsewhere, though (like in HomeAssistant).

Going to close since I don't think there's a good way to handle this within the project itself, but appreciate the idea and suggestion.

closed 2019-10-05 by sidoh