sidoh/esp8266_milight_hub #529: White Modus

poudenes, 2019-09-25:
1.10.3 (nodemcuv2)

When select White my RGB+CCT bulbs gives Kelvin color not white

When select Night mode and then select White nothing happen
When select Night mode and then select a color nothing happen
Turn off and on again will break this problem them you can select White or a color again.

sidoh, 2019-09-27:
I think I expect this behavior.  Night mode is a special "status" like on or off.  This is actually how the bulbs behave.  If they're in night mode, they don't react to any commands other than "on" or "off".  I could paper over this behavior in the firmware, but I'm inclined to steer clear of this sort of thing because state management gets unwieldy really quickly.

Given this, I'm going to close, but please let me know if I've misunderstood.

closed 2019-09-27 by sidoh

poudenes, 2019-09-27:
Thanks for the clear answer :)