sidoh/esp8266_milight_hub #530: Can't save settings.

mpisek, 2019-09-25:
So, I'm using NodeMCU for more than I year now. But in last month it stopped working. I re-flashed it with 1.10.3 but now I can't save settings. When I refresh my browser or reset the device the settings are gone. Wrote-Over all 4MB with blank 1MB file and re-flashed with 1.10.0 still the same story. It saves the initial wifi config SSID and password but that's about all. After reboot I can open the WebGUI but settings can't be saved. What am I missing?

sidoh, 2019-09-25:
Hi, please check serial output when saving settings.  What version were you using when it stopped working?

mpisek, 2019-09-25:
Last working for me was 1.10.0-rc.3 (nodemcuv2).
When I open the web GUI I get serial output:
Unhandled websocket event: 10
Opening settings file failed

sidoh, 2019-09-26:
If you revert back to 1.10.0-rc.3, everything works?  Given the changelog, that could basically only mean something in the toolchain silently changed.

mpisek, 2019-09-28:
No. Same thing. Nothing works even on 1.10.0-rc.3 not even on earlier versions. I ordered a new NodeMCU but I have the same problem on the new hardware. Now I'm extra confuzed.

sidoh, 2019-09-28:
can you paste the output of /about?

sidoh, 2019-09-28:
i'm confused too, that makes no sense to me :S

sidoh, 2019-09-28:
definitely try running `esptool.py erase_flash` before flashing the firmware.

mpisek, 2019-09-28:
Just did that.

mpisek, 2019-09-28:
▒▒5▒▒▒Opening settings file failed
*WM: Adding parameter
*WM: staticIP
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.125.107
Wifi connected succesfully

Setup complete (version 1.10.3)

mpisek, 2019-09-28:
Let me try with https://www.espressif.com/en/support/download/other-tools different USB cable etc.

mpisek, 2019-09-28:
Hmm, it didn't work with expressif - Flash Download Tools (ESP8266 & ESP32) either. What did work was  nodemcu-flasher. I don't get it. But hey it works :)

sidoh, 2019-10-05:
That's really confusing.  There are other users reporting similar problems, so I'm pretty sure there's something funky going on.  This is helpful.

Can you paste a screenshot of the settings you used to flash with nodemcu-flasher?  Did you flash it with a pre-compiled binary?

mpisek, 2019-10-07:
I used precompiled library yes. Well with (Flash Download Tools (ESP8266 & ESP32)) I tryed allkind of settings and nothing worked really. With nodemcu-flasher default settings worked.

jkhunter, 2019-10-17:
Pleas try another browser. I have same issue with Waterfox 56.2.14 (64-bit) on linux. I open the webui in chrome Version 77.0.3865.90 (Official Build) (64-bit) and saving worked for me.

jflemieux, 2019-12-02:
Same workaround for me. Restoring does not work on Firefox.

On Chrome after the Restore, I have to re-access /settings to complete the restore.

jellewie, 2020-01-07:
@mpisek you could try one of the 3 options I mentioned in #584  as I did also get the "Opening settings file failed" error when using bin uploaders if the Flash size was not set properly. But managed to get it working in the end.

Also use "Shift+F5" on the ESP webpage, this is a force clean reload (it won't use old bad stuff from the browser memory), but this cant be the cause of the error since "Opening settings file failed" is an internal ESP error that it could not get the data from flash

Tontze, 2020-05-27:
Same problem here, im using wemos d1 mini, precompiled binary. Using tasmotizer for flashing. Tasmotizer worked well for me earlier, now my network has changed, and i cannot save any settings .. Not on firefox, ie or chrome. IE does not even ask password after setting it.

didi-home, 2021-01-14:
I've just assembled my hub this evening and also experienced the issue that settings could not be saved.
However, for me the PUT request for saving always returned "success" but the GET /settings ran into connection reset.

I tried flashing 2x with Tasmotizer (been using that a lot lately for ES8266 devices without any problems) and I had this issue both times. Finally decided to download nodemcu-flasher (https://raw.githubusercontent.com/nodemcu/nodemcu-flasher/master/Win64/Release/ESP8266Flasher.exe) and now everything works!

So, yes, there is something fishy in the flash process - @sidoh maybe you can put nodemcu-flasher into the README.md as recommended tool?