sidoh/esp8266_milight_hub #531: Define specific http port

agoralive, 2019-09-26:
To make the MDNS better on windows computers I would like to change the default webservice port from 80 into something else.

I have no problem gaining access to my Node-Red on raspberry.local:1880 on windows, but the milight-hub.local is not working (MAC works fine.....)

SO, could I change the port 80 to (I don't know) 12345 instead to make windows compatible?

sidoh, 2019-09-26:
I doubt the port is the issue.  You could try changing it manually if you like.

https://github.com/sidoh/esp8266_milight_hub/blob/master/src/main.cpp#L414
https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/WebServer/MiLightHttpServer.h#L35

sidoh, 2019-09-29:
Please let me know if changing the port helps.

Setting for non-default HTTP port is also a duplicate of #183, so going to close.

closed 2019-09-29 by sidoh

agoralive, 2019-12-11:
Changing ports did not help, Windows stinks in this matter, Android systems does as well. Only sure bet for MDNS is the MAC/IOS devices