sidoh/esp8266_milight_hub #532: Bulbs not react

poudenes, 2019-09-27:
Hi,

Since the change to 1.10.0 till latest one 1.10.3 I see some problems with the firmware.
The firmware build 1.9.2 all bulbs react on the signal that has been send to it.

Now sometimes 1 bulb random in the house does not react on the signal.
All settings are the same all the time.

I can try to revert back to 1.9.2 but then I have to change my HA also, have some transition actions add in yaml script. But then I can see if this problem is gone or not.

sidoh, 2019-09-27:
Would be really helpful if you find a consistent way to reproduce the issue you're seeing, as I've been using 1.10 in my setup since the early days without any problems.  

Please also provide the info asked for in the issue template, especially the output of `/settings` and `/about`.

https://github.com/sidoh/esp8266_milight_hub/issues/new?assignees=&labels=bug&template=bug_report.md&title=

Serial logs would also be helpful.

poudenes, 2019-10-01:
Hi Sidoh, 

Sorry I didn't had some time to do some tests. Will come back on this one...

poudenes, 2019-10-09:
I was try to do some tests. But my partner get grazy now and want to stop me do all the things. :D

So for now I ordered a module to test things on a separate hub. Want test on that one then.
In Meanwhile  I want to know if it will help to use the QoS Setting in HA change to 1 or even 2

Have a very stable Wifi Connection.

sidoh, 2019-10-12:
No, QoS shouldn't matter.

This seems like it's probably the same issue you're discussing in #539, right?

poudenes, 2019-10-19:
I run now latest version again. Change the 
Packet Repeats to 50
Packet Repeats per loop to 50
Packet Repeat throttle threshold to 500
Packet Repeat throttle Sensitivity to 10
Packet Repeat Minimum to 50

Seems this solved the problem. But ill keep a eye on it.

closed 2019-10-19 by poudenes