sidoh/esp8266_milight_hub #533: No Commands are being sent, but Sniffing and Receiving works.

bennythestang, 2019-09-30:
Did a normal Setup yesterday, with esp8266/NodeMCUv3 & nRF24L01+PA+LNA.
Everything works fine except **sending Signals wirelessly**. But Sniffing and Receiving Signals (from existing MiLight Remote) works without any Problems.

I did a checkup with 2 Setups (One for Sending, One for Sniffing)  to verify the Problem. Doublechecked everything else, but did not find any error.

Is there a known Issue? Did i overlook something? What can i do to analyse the Problem?

Many Thanks in Advance!
Benny

sidoh, 2019-10-05:
Hi @bennythestang,

If you haven't already, make sure you check out the [troubleshooting page](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting).

Wanted to make sure I understood this:

> I did a checkup with 2 Setups (One for Sending, One for Sniffing, then switched, the problem exists on both setups!) to verify the Issue. Doublechecked everything else, but did not find any error.

Do you mean to say that neither setup detected packets from the other?

If so, yes, this can happen.  The best suggestions I have for this are:

1. Try moving the nRF24 further away from the ESP8266.  They're both 2.4 GHz and interference can be a real problem.  Relatedly, try putting the sender very close to the receiver (shouldn't be necessary, but can help rule things out)
1. Make sure you're using a good power supply (laptop USB should work great).
1. Try another nRF24.  They build quality does not seem to be terribly consistent.
1. Try a capacitor (~10µF) across Gnd and Vin on the nRF.

bennythestang, 2019-10-05:
Hi @sidoh ,

thanks for your reply! First of all, i fixed the main issue but after that i ran into another more confusing Problem ;-)

To get signals and commands sent wirelessly i did the following, as mentioned in Issue #449 :
I connected "CE" to "D4" on the NodeMCUv3, and in the setting set "2" for the GPIO (as D4 is GPIO2). This works now!

To verify this, i received and sniffed alle signals with my second setup and everythings works fine,
**BUT no MiLight Devices are responding** (LS2, FUT013, FUT015).

I will make a Video to show that behaviour.

sidoh, 2019-10-05:
Do the devices respond to a real milight remote?

#498 (and [this comment in particular](https://github.com/sidoh/esp8266_milight_hub/issues/498#issuecomment-522237261)) might be relevant.  This was a super strange issue that I still don't understand.  But swapping the nRF24 fixed it.

bennythestang, 2019-10-05:
Yes the Devices are responding to the original milight remotes.

I can sniff the Packets form the original remote and spoof the ID. With my second NodeMCU/nRF24 Setup (always there to verify my setup) i can sniff the packets form the first NodeMCU/nRF24 Setup and the original remote. The Sniffing Log looks fine. But the MiLight Devices are only responding to the original Remote.

sidoh, 2019-10-05:
What results are inconsistent?  

If you're only sometimes receiving packets, I'd suggest:

1. Making sure sure the wiring is secure.  Switch to soldered connections if at all possible.
1. Make sure you're using a solid power supply

if all else fails, i'd try ordering new nRFs.  the ones you have might be from the same batch.

sidoh, 2019-11-16:
Going to close due to inactivity.  Feel free to bump the issue and we can reopen (keep in mind, though, that I can't be of much help if I can't reproduce the issue).

closed 2019-11-16 by sidoh