sidoh/esp8266_milight_hub #537: No remote reaction after weeks

boaschti, 2019-10-04:
### Describe the bug

No reaction after weeks when any button on a remote controll is pressed. After rebooting the device the device works fine again. The device is reachable in the network. It seems that the NRF has lost its configuration and gets no radio data.
Is it possible to reinit the NRF without reboot to contain the problem?

### Steps to reproduce

I cannot reproduce it. Some times it works for months sometimes only one week.

### Expected behavior

### Setup information
WEMOS D1 mini with NRF24

#### Firmware version
1.10.1 (d1_mini) Same beaviour with previous versions.

#### Output of http://milight-hub.local/about

```{"firmware":"milight-hub","version":"1.10.1","ip_address":"192.168.178.26","reset_reason":"Software/System restart","variant":"d1_mini","free_heap":17616,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```{"admin_username":"peep","admin_password":"peep","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"peep","mqtt_username":"peep","mqtt_password":"peep","mqtt_topic_pattern":"MiLight/:device_id/:group_id","mqtt_update_topic_pattern":"MiLight2/:device_id/:group_id","mqtt_state_topic_pattern":"MiLight3/:device_id/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","rf24_channels":["LOW","MID","HIGH"],"device_ids":[],"gateway_configs":[],"group_state_fields":["state","brightness","level","saturation","mode","color_temp","computed_color"],"group_id_aliases":{}}
```

### Additional context

sidoh, 2019-10-05:
I sort of doubt the nRF is the real source of the problem.  Seems more likely that the ESP8266 is getting into a weird state.  When this is happening next time, can you capture `/about` and also try to poke around on the webpage?  See if sending/listening works.

Serial logs would also be interesting, but requires you have the ESP8266 plugged into a device that can read them at all times 🤐.

Not a ton I can do about this without being able to reproduce it (and I haven't noticed anything like this in my setup).  You _can_ set up your hub to reboot on a schedule, in case that's helpful.  It's under the "setup" tab in settings.

RURon, 2019-10-22:
Hm, I see the same behaviour here with a FUT092 remote. At first I thought I did something wrong as I did not receive any events in the sniffer. But after rebooting the ESP, I now see them. As my ESP was not powered for longer than 24 hours, I wonder when this will happen again as I would like to use some MiLight remotes to control other things in my home (than Milights)...

boaschti, 2019-11-01:
Hi, the mi light stuck again. 

/about:

> {"firmware":"milight-hub","version":"1.10.1","ip_address":"192.168.178.26","reset_reason":"Software/System restart","variant":"d1_mini","free_heap":17152,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}

If i start sniffing on web page i see that no packets are received. There are no problems with the web server. 
I'll try tomorrow to read the serial without disconnecting and reboot. Is a information what is happening on spi also interesting for you. I can record it with a logic analyser.
Best regards

sidoh, 2019-11-02:
Alright, interesting.  Yes, Serial logs would be interesting.  The `/about` output at least indicates that there are no crashes.

boaschti, 2019-11-02:
Ok. I'll log it tomorrow.

sidoh, 2019-11-16:
Any luck here?

boaschti, 2019-11-16:
No. At the moment it works fine. I don't know if esp reboots itself or reboot happend.

LEJOUI, 2019-11-22:
So, it’s known that esp8266 sometimes stop working. I also encountered this with the nodemcu several times now. Only solution is to restart it. The board can’t be accesed at that moment. Perhaps it’s an idea to make it an option that the board automatically reboots every day or 7 days e.g.?

garmck, 2019-11-24:
> So, it’s known that esp8266 sometimes stop working. I also encountered this with the nodemcu several times now. Only solution is to restart it. The board can’t be accesed at that moment. Perhaps it’s an idea to make it an option that the board automatically reboots every day or 7 days e.g.?

@LEJOUI its already an option. Settings\Setup\Auto Restart Period

LEJOUI, 2019-11-26:
> @LEJOUI its already an option. Settings\Setup\Auto Restart Period

Thank you!

sidoh, 2019-11-30:
I'm going to close since it doesn't seem like this is a persistent problem.  Obviously happy to re-open if it happens again.

closed 2019-11-30 by sidoh