sidoh/esp8266_milight_hub #538: Settings get lost

poudenes, 2019-10-09:
### Describe the bug

Within seconds no lights worked anymore. Checked Milight hub and settings were set disable

[image]

### Steps to reproduce

Cannot reproduce. I use 28 bulbs in HA. HA sends MQTT signals to the hub

### Setup information

#### Firmware version

1.10.3 (nodemcuv2)

#### Output of http://milight-hub.local/about

```
{"firmware":"milight-hub","version":"1.10.3","ip_address":"192.168.100.70","reset_reason":"External System","variant":"nodemcuv2","free_heap":14328,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":25,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.100.24:1883","mqtt_username":"XXXXX","mqtt_password":"XXXXX","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":100,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":500,"packet_repeat_minimum":10,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.68.100.70","wifi_static_ip_gateway":"192.168.100.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"],"group_id_aliases":{"Balcony":["rgb_cct",43684,1],"Bathroom":["rgb_cct",43693,1],"Dressoir":["rgb_cct",43695,1],"Hallway":["rgb_cct",43683,1],"Kitchen":["rgb_cct",43682,1],"Livingroom":["rgb_cct",43692,1],"MasterBedroom1":["rgb_cct",43690,1],"MasterBedroom2":["rgb_cct",43691,1],"MasterBedroomAll":["rgb_cct",43706,1],"Sink":["rgb_cct",43681,1],"Toilet":["rgb_cct",43694,1],"WallWindow":["rgb_cct",43707,1]}}
```

### Additional context
With this version also sometimes bulbs not go off. (already post this)

sidoh, 2019-10-09:
Thanks for reporting.  This is probably a duplicate of #523.  Looking likely that the issue is a memory leak, although it's not yet clear where it's coming from.

Going to close so that we can keep the discussion centralized.  I'll tag you on the issue with some questions.

closed 2019-10-09 by sidoh

errrrata, 2020-06-29:
Sorry for reopening this, but I just experienced the same issue, after months of gateway working without any issue.
I'm running 1.10.6, combined with HA, and today I had to restart the ESP8266 as it wasn't responding to ping. It came back online, so the wifi settings are still saved, however checking the interface it show all the settings empty and the same goes for all the saved remotes. HA also reports now the lights as unavailable.

I'm not sure if settings are really getting lost, since the wifi connection seems to be kept?

sidoh, 2020-07-01:
Hey @errrrata, thanks for bumping this. The ticket for the memory leak issue is #601.

Wifi settings are stored in their own region of flash. Settings are stored on SPIFFS. I think the leak is somehow causing the SPIFFS partition table to get corrupted, but haven't verified that yet.

errrrata, 2020-09-20:
Hi @sidoh. First of all, thank you for your reply.

I think I found a way to reproduce this issue, in case it can be any help in solving it.
The issue in my case seems to be adding too many group_id_aliases (more than 18 in my case).

So to give u a little bit of background, as I mentioned above, I have tested the gateway for few months, everything worked pretty well, so I decided to add more devices. Whenever I do this, everything works as expected, I add all my devices, start setting them up in Home Assistant, and after a while when I check back in milighthub, everything is gone. To overcome this issue, I started making backups after adding new devices, and I was surprised to discover that at some point, the backups cannot be restored. They are restored, it gives me the "{"success": true}", but going back to the web interface still shows empty. So I decided to modify the backup file manually and see which part makes it fail. 
My first idea was that it might be something memory related, so I shortened the names of the group_id_aliases greatly, but with no success. What seemed to work however was reducing the group_id_aliases to 18, once I get to this number, a restore works, once I add another item (I tried various combinations doesn't seem to be related to which item but the count of items), restore says it's ok, but when reloaded the hub is empty.
I haven't yet tried to see if less "device_ids" makes any difference, as I currently have 5 of them
If needed, I can help debugging more in this issue, but I wanted to tell u what I find as I saw in previous discussions about this 'random resets' that the difficulty is in replicating this issue

poudenes, 2020-09-20:
@errrrata maybe you can do something with my settings? I use almost 35 bulbs in 12 virtual remotes:

```
{
	"admin_username": "",
	"admin_password": "",
	"ce_pin": 16,
	"csn_pin": 15,
	"reset_pin": 0,
	"led_pin": -2,
	"radio_interface_type": "nRF24",
	"packet_repeats": 50,
	"http_repeat_factor": 1,
	"auto_restart_period": 0,
	"mqtt_server": "192.168.100.XXX:1883",
	"mqtt_username": "XXXXX",
	"mqtt_password": "XXXXX",
	"mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
	"mqtt_update_topic_pattern": "milight/updates/:hex_device_id/:device_type/:group_id",
	"mqtt_state_topic_pattern": "milight/states/:hex_device_id/:device_type/:group_id",
	"mqtt_client_status_topic": "milight/client_status",
	"simple_mqtt_client_status": true,
	"discovery_port": 48899,
	"listen_repeats": 3,
	"state_flush_interval": 200,
	"mqtt_state_rate_limit": 500,
	"mqtt_debounce_delay": 500,
	"packet_repeat_throttle_sensitivity": 10,
	"packet_repeat_throttle_threshold": 500,
	"packet_repeat_minimum": 50,
	"enable_automatic_mode_switching": false,
	"led_mode_wifi_config": "Fast toggle",
	"led_mode_wifi_failed": "Fast blip",
	"led_mode_operating": "Slow toggle",
	"led_mode_packet": "Fast blip",
	"led_mode_packet_count": 4,
	"hostname": "MilightHub",
	"rf24_power_level": "MAX",
	"rf24_listen_channel": "LOW",
	"wifi_static_ip": "192.68.100.xxx",
	"wifi_static_ip_gateway": "192.168.100.1",
	"wifi_static_ip_netmask": "255.255.255.0",
	"packet_repeats_per_loop": 50,
	"home_assistant_discovery_prefix": "",
	"wifi_mode": "g",
	"default_transition_period": 500,
	"rf24_channels": ["LOW", "MID", "HIGH"],
	"device_ids": [43681, 43682, 43683, 43684, 43690, 43691, 43692, 43693, 43694, 43695, 43706, 43707],
	"gateway_configs": [],
	"group_state_fields": ["state", "status", "brightness", "level", "hue", "saturation", "color", "mode", "kelvin", "color_temp", "bulb_mode", "computed_color", "effect", "device_id", "group_id", "device_type"],
	"group_id_aliases": {
		"bc": ["rgb_cct", 43684, 1],
		"br1": ["rgb_cct", 43690, 1],
		"br2": ["rgb_cct", 43690, 2],
		"br3": ["rgb_cct", 43690, 3],
		"br4": ["rgb_cct", 43690, 4],
		"br5": ["rgb_cct", 43691, 1],
		"br6": ["rgb_cct", 43691, 2],
		"bt0": ["rgb_cct", 43693, 0],
		"bt4": ["unknown", 0, 0],
		"ds0": ["rgb_cct", 43695, 0],
		"ds1": ["rgb_cct", 43695, 1],
		"ds2": ["rgb_cct", 43695, 2],
		"ds3": ["rgb_cct", 43695, 3],
		"hl": ["rgb_cct", 43683, 1],
		"kc0": ["rgb_cct", 43682, 0],
		"kc1": ["rgb_cct", 43682, 1],
		"kc2": ["rgb_cct", 43682, 2],
		"kc3": ["rgb_cct", 43682, 3],
		"lr0": ["rgb_cct", 43692, 0],
		"lr1": ["rgb_cct", 43692, 1],
		"lr2": ["rgb_cct", 43692, 2],
		"lr3": ["rgb_cct", 43692, 3],
		"lr4": ["rgb_cct", 43692, 4],
		"sk0": ["rgb_cct", 43681, 0],
		"sk1": ["rgb_cct", 43681, 1],
		"sk2": ["rgb_cct", 43681, 2],
		"sk3": ["rgb_cct", 43681, 3]
	}
}
```

errrrata, 2020-09-20:
One major difference I see between your settings and mine, you're not using 'group_id_aliases' which is exactly what causes issues for me.

DreamView, 2020-11-01:
I have the exact same issue, when I configure more 'group_id_aliases' the setting get lost and I cannot restore from backup anymore. I'm running in a Wemos D1 mini. A solution would be great, because the aliases are very handy when pairing with Web UI

sidoh, 2020-11-01:
Hi @DreamView, the issue you're referring to is pretty well understood, but fixing will take a level of time and effort I don't have to give to the project right now.

The relevant open issue is #577.

Basically settings are saved as a monolithic JSON blob, which is artificially limited to a few KBs. It's not terribly complicated to split out the parts of settings that are expandable, but will be a fairly large undertaking.