sidoh/esp8266_milight_hub #539: Leds not react

poudenes, 2019-10-10:
### Describe the bug

The led signals not working. Wifi connected, led when packages received.

[image]

### Steps to reproduce

After update to 1.10.x

### Expected behavior

Must work, to see if Wifi is connected, or packaged are received

### Setup information

#### Firmware version
1.10.3 (nodemcuv2)

#### Output of http://milight-hub.local/about

```
{"firmware":"milight-hub","version":"1.10.3","ip_address":"192.168.100.70","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":14008,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":25,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.100.24:1883","mqtt_username":"XXXXX","mqtt_password":"XXXXX","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":100,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":500,"packet_repeat_minimum":10,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Fast blip","led_mode_packet_count":3,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.68.100.70","wifi_static_ip_gateway":"192.168.100.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"],"group_id_aliases":{"Balcony":["rgb_cct",43684,1],"Bathroom":["rgb_cct",43693,1],"Dressoir":["rgb_cct",43695,1],"Hallway":["rgb_cct",43683,1],"Kitchen":["rgb_cct",43682,1],"Livingroom":["rgb_cct",43692,1],"MasterBedroom1":["rgb_cct",43690,1],"MasterBedroom2":["rgb_cct",43691,1],"MasterBedroomAll":["rgb_cct",43706,1],"Sink":["rgb_cct",43681,1],"Toilet":["rgb_cct",43694,1],"WallWindow":["rgb_cct",43707,1]}}
```

### Additional context

poudenes, 2019-10-10:
When I went closer to the board. I see a very very very weak light. Its working but its almost off.

sidoh, 2019-10-10:
Seems unlikely that this is due to a version upgrade if it works at all on 1.10.  Please try reverting to 1.9.

poudenes, 2019-10-11:
Revert back to 1.09.2.
I test this problem to turn bulbs on and off all the time. With 1.10.x after 3 or 4 times turn on and off there is 1 bulb that stay on. With 1.09.2 it takes me 14 times before the first bulb also stay on.

This is maybe not the best way to test, but for me there is no other option.
( All DIY modules are behind plexiglass where electricity comes into the house)

[image]

sidoh, 2019-10-11:
25 repeats is pretty low (mine is set to 50).  Have you tried increasing this?  

Guess it could be a timing issue.  By default, repeat sends are handled in groups of 10.  You can also try setting "number of repeats per loop" equal to the number of repeats, which should cause the hub to mostly mimic the packet behavior in 1.9. 

But really, if bulbs are missing commands at all, it means your repeat count is too low.

poudenes, 2019-10-12:
I went back to 1.10.3 and tried it. But then when I use KaKu switch and turn on and off within 5 times a bulb start on. On 1.9.2 it takes more then 25 when this will happen.

Have change the "number of repeats per loop" equal to number of repeats in 1.10.3 before I test it.

sidoh, 2019-10-12:
> But then when I use KaKu switch and turn on and off within 5 times a bulb start on. On 1.9.2 it takes more then 25 when this will happen.

Sorry, I'm not following what you mean here.  Can you clarify?

sidoh, 2019-10-12:
Ah, nevermind.  I think I understand.

Honestly not sure I'm gonna be able to troubleshoot much here.  If you end up finding the issue yourself, happy to look into it.  But:

1. I'm not able to replicate this locally, so really difficult for me to make progress myself
2. The setting I suggested _should_ provide the same behavior as packet sending in 1.9

as a workaround, the best bet is probably just to increase the number of repeats.

poudenes, 2019-10-12:
What antenna do you have?

sidoh, 2019-10-12:
I'm using the standard nRF24L01+ with a PCB antenna.

poudenes, 2019-10-12:
I use this one:

https://www.hobbyrc.co.uk/frsky-t-dipole-antenna-ipex-1-868mhz?gclid=EAIaIQobChMI1bruipqX5QIVQuh3Ch11CgElEAQYAiABEgLKzPD_BwE

I can try to use a other one. But this one must be bettet

sidoh, 2019-10-12:
I don't think it's a hardware problem.  

I tried swapping between 1.10 and 1.9 with different repeat counts and wasn't able to reliably reproduce this problem.

I have pretty limited capacity to work on this project these days, and troubleshooting these sorts of issues takes a ton of time without a way to reliable way to reproduce the problem.

sidoh, 2019-10-12:
You'll almost certainly get better reliability results if you turn up the number of repeats to, say, 50.  What concerns do you have with doing this?

poudenes, 2019-10-12:
Let me try that tomorrow. will change to 50. And try it with latest firmware.
Ill come back with the results

poudenes, 2019-10-16:
I run now latest version again. Change the 
Packet Repeats to 50
Packet Repeats per loop to 50
Packet Repeat throttle threshold to 500
Packet Repeat throttle Sensitivity to 10
Packet Repeat Minimum to 50

Seems this solved the problem. But ill keep a eye on it.

beside this transitions have strange behaviour now.
When brightness to up from 20 to 100 and send action to 2 groups the bulbs are flickering 
light up brightness up, down, more up, down again, more up etc... both groups doing this.
(ill do some tests to see what this can be)

poudenes, 2019-10-21:
seems to work.

closed 2019-10-21 by poudenes