sidoh/esp8266_milight_hub #540: [Question] Color Temperature of Light differs from my milight dimmer

fluffymadness, 2019-10-12:
I have bought one of these fut035 cw ww dual channel dimmers that have a range from 2700-6500k and I put a 24v constant voltage led panel behind it.
The thing is the color range of my panel goes only from 3000k to 5000k
Manually dimming works completely fine with the remote and your milight hub.
Thing is when using addons like fluxer or circadian for homeassistant that want to set an exact color temperature things are getting more interesting.
Since e.g the exact 3000k it wants to have is never 3000k on the lightsource, same goes for the 4000k setting during daytime.
Do you maybe have some ideas how i could work around that issue?

sidoh, 2019-10-12:
I'm not familiar with how fluxer or circadian work, so not going to be of much direct help.

espMH does expose the ability to set the color temp in mireds, but the min and max values were determined from the specs of one bulb type and is almost certainly not correct for other types.

fluffymadness, 2019-10-12:
Ah ok. Fluxer or Circadian just sends the mireds it wants to have via mqtt.
Basically I have no direct control over the color temperature that is set, because it changes automatically.
Well there dies the hope of a quick fix on the hubs side.
I would probably have to write some code that does some sort of mapping from one scale to another to map the 3000-5000k range to 2700k-6500k to get it working correctly.
I think I'll do some research on the homeassistant-side then.

closed 2019-10-12 by fluffymadness

sidoh, 2019-10-12:
👍 sounds good.

The underlying milight protocol has no information about the true color temperature it's setting.  It's just a range from 0-100.

garmck, 2019-10-12:
@fluffymadness you can customize your lights in homeassistant to set the max mired value.
For milight i use 370 as that value.
You can adjust this to suit your needs.