sidoh/esp8266_milight_hub #542: Unable to create a transition with firmware 1.10.3

Chrominator, 2019-10-14:
When issuing : 
```
curl -vvv -u Chrominator http://milight-hub.local/transitions -X PUT \
        --header "Content-Type: application/json" \
        --data-binary '{ "field": "brightness", "start_value":0, "end_value":100, "duration":5, "period":50, "id":0, "last_sent":0,
"bulb": { "device_id":0x8B6C, "group_id":1, "device_type": "cct" }, "current_value":0 }'
```

got the following : 

```
Enter host password for user 'Chrominator':
*   Trying 192.168.1.11...
* TCP_NODELAY set
* Connected to milight-hub.local (192.168.1.11) port 80 (#0)
* Server auth using Basic with user 'Artige'
> PUT /transitions HTTP/1.1
> Host: milight-hub.local
> Authorization: Basic QXJ0aWdlOkNlbnR1cnkyMQ==
> User-Agent: curl/7.58.0
> Accept: */*
> Content-Type: application/json
> Content-Length: 196
>
* upload completely sent off: 196 out of 196 bytes
< HTTP/1.1 404 Not Found
< Content-Type: text/html
< Content-Length: 23
< Connection: close
<
* Closing connection 0
Not found: /transitions
```
I§ don't understand what's wrong in my syntax, the device is working well with the web based interface or with other REST APs.
Thank you for your help.

`http://milight-hub.local/transitions` is producing the following output : 
`{"transitions":[]}`

About : 
`{
  "firmware": "milight-hub",
  "version": "1.10.3",
  "ip_address": "192.168.1.11",
  "reset_reason": "Software/System restart",
  "variant": "nodemcuv2",
  "free_heap": 16312,
  "arduino_version": "2_4_2",
  "queue_stats": {
    "length": 0,
    "dropped_packets": 0
  }`

Settings : 

```
{
  "admin_username": "Chrominator",
  "admin_password": "",
  "ce_pin": 4,
  "csn_pin": 15,
  "reset_pin": 0,
  "led_pin": -2,
  "radio_interface_type": "nRF24",
  "packet_repeats": 50,
  "http_repeat_factor": 1,
  "auto_restart_period": 1440,
  "mqtt_server": "192.168.1.4:1883",
  "mqtt_username": "",
  "mqtt_password": "",
  "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
  "mqtt_update_topic_pattern": "milight/updates/:hex_device_id/:device_type/:group_id",
  "mqtt_state_topic_pattern": "milight/states/:hex_device_id/:device_type/:group_id",
  "mqtt_client_status_topic": "",
  "simple_mqtt_client_status": false,
  "discovery_port": 48899,
  "listen_repeats": 3,
  "state_flush_interval": 10000,
  "mqtt_state_rate_limit": 500,
  "packet_repeat_throttle_sensitivity": 0,
  "packet_repeat_throttle_threshold": 200,
  "packet_repeat_minimum": 3,
  "enable_automatic_mode_switching": false,
  "led_mode_wifi_config": "Fast toggle",
  "led_mode_wifi_failed": "On",
  "led_mode_operating": "Slow blip",
  "led_mode_packet": "Flicker",
  "led_mode_packet_count": 3,
  "hostname": "milight-hub",
  "rf24_power_level": "MAX",
  "rf24_listen_channel": "LOW",
  "wifi_static_ip": "192.168.1.11",
  "wifi_static_ip_gateway": "192.168.1.251",
  "wifi_static_ip_netmask": "255.255.255.0",
  "packet_repeats_per_loop": 10,
  "home_assistant_discovery_prefix": "",
  "wifi_mode": "n",
  "default_transition_period": 500,
  "rf24_channels": [
    "LOW",
    "MID",
    "HIGH"
  ],
  "device_ids": [
    8294,
    16280,
    16281,
    33428,
    35692
  ],
  "gateway_configs": [
    [
      33428,
      5987,
      5
    ],
    [
      16280,
      5988,
      5
    ],
    [
      16281,
      5989,
      5
    ],
    [
      35692,
      5990,
      5
    ]
  ],
  "group_state_fields": [
    "state",
    "brightness",
    "level",
    "hue",
    "saturation",
    "color",
    "mode",
    "kelvin",
    "color_temp",
    "bulb_mode",
    "computed_color",
    "effect"
  ],
  "group_id_aliases": {
    "Bureau": [
      "rgbw",
      16281,
      1
    ],
    "Couloir": [
      "cct",
      35692,
      1
    ],
    "Salon": [
      "rgbw",
      16280,
      2
    ]
  }
}
```

sidoh, 2019-10-14:
The correct verb is `POST`, not `PUT`.  Check out the documentation here:

https://sidoh.github.io/esp8266_milight_hub/branches/latest/#tag/Transitions/paths/~1transitions/post

closed 2019-10-14 by sidoh

Chrominator, 2019-10-14:
Sorry.
Was refering to a page : [https://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/](url) on "Using the REST API" section.

sidoh, 2019-10-15:
Ah, I'm sorry.  It's possible the API docs are broken.  I'll take a look later.

The easiest way to schedule a transition is probably just by including the `"transition"` key in a command, so:

```
curl -X PUT -H 'Content-Type: application/json' -d '{"brightness":255,"transition":60}' 192.168.1.11/gateways/0x8B6C/cct/1
```

Chrominator, 2019-10-15:
Really good job, thank you again !
I like this improvement.