sidoh/esp8266_milight_hub #545: Lamps turn red

pitah81, 2019-10-31:
Version: 1.10.3

Unfortunately I don't recall which version I was running before. But after upgrading to 1.10.3 I have a problem with my RBGW lamps turning red whenever I do changes in the settings or when the controller looses wifi and reconnects. Why would this happen all of a suddenly? There's no rules in Home Assistant to change lamps into red and no Node Red flows either. Just by restarting the ESP controller all lamps turn red except for my newer CCT/RGBW spotlight lamps.

poudenes, 2019-11-01:
This is not only with 1.10.3. I have this since begin. For me I don't care this happen. Turn on bulb and turn off in HA. Then everything is ok then :)

I don't know if this is a bug or not haha ... :D

sidoh, 2019-11-02:
Are you using MQTT?  Seems like you might have some retained command messages that are doing this.  Try digging around with `mosquitto_sub`.  You can erase retained messages by publishing a blank message to the same topic.

pitah81, 2019-11-02:
Thanks @sidoh, it appears I had retained command in my mosquitto. I emptied the mosquitto database and now my lamps stay white. I just don't know where it got the retained red command from, I can't recall ever sending red to all my lamps, must be a bug in HA or something else.

closed 2019-11-02 by pitah81

poudenes, 2019-11-03:
Beside this solution. Is it ok to use retain in HA, or remove that line?
`retain: true`
I read positive things and negative things of using it.