sidoh/esp8266_milight_hub #546: FUT091 produces "night_mode" commands when pressing On/Off buttons longer

RURon, 2019-11-01:
### Describe the bug

I try to use a FUT091 to control a music player. The remote has single ON and OFF buttons for Group 0 - when I press either of these only shortly (and am within range of the ESP), then I successfully get the expected Group 0 ON or OFF command via MQTT. Unfortunately I often have to press either button longer to get a command thru to the receiver. And this results in a series of "night_mode" messages being sent - disregarding whether I'm pressing the ON or OFF button on the remote.

Therefore I often cannot get a useful command via MQTT. Also the other channel's on/off buttons show the same behaviour.

When the radio link is good, I can see in the logs that I am receiving a correct message as first message and afterwards only these "night_mode" messages.
Sniff result on short button press

```
fut091 packet received (9 bytes):
Raw packet: 61 55 38 A3 19 F4 5B 1B 9B 

Decoded:
Key      : 61
b1       : 21
ID       : 7853
Command  : 01
Argument : 05
Sequence : 94
Group    : 00
Checksum : 9F
```
When pressing the same key for, say 2 seconds, I get this:

```
fut091 packet received (9 bytes):
Raw packet: CA E1 77 49 8A D9 BE C9 25 

Decoded:
Key      : CA
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : A1
Group    : 00
Checksum : 01

fut091 packet received (9 bytes):
Raw packet: F9 BD 90 4B 41 9C F7 C3 7F 

Decoded:
Key      : F9
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : A0
Group    : 00
Checksum : D3

fut091 packet received (9 bytes):
Raw packet: DC F8 15 2E C2 B1 11 41 98 

Decoded:
Key      : DC
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : 9F
Group    : 00
Checksum : A5

fut091 packet received (9 bytes):
Raw packet: 53 A0 2E 61 D7 31 3D 6C 76 

Decoded:
Key      : 53
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : 9E
Group    : 00
Checksum : 77

fut091 packet received (9 bytes):
Raw packet: 76 2D 23 F5 D6 2D 16 15 A9 

Decoded:
Key      : 76
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : 9D
Group    : 00
Checksum : 49

fut091 packet received (9 bytes):
Raw packet: 65 49 3C B7 8D F0 6F 0F 23 

Decoded:
Key      : 65
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : 9C
Group    : 00
Checksum : 1B

fut091 packet received (9 bytes):
Raw packet: 98 44 D1 FA 0E F5 49 8D 94 

Decoded:
Key      : 98
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : 9B
Group    : 00
Checksum : ED

fut091 packet received (9 bytes):
Raw packet: FF EC DA 0D 23 85 6D B8 72 

Decoded:
Key      : FF
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : 9A
Group    : 00
Checksum : BF

fut091 packet received (9 bytes):
Raw packet: F2 B9 DF 61 22 71 8E 61 4D 

Decoded:
Key      : F2
b1       : 21
ID       : 7853
Command  : 81
Argument : 05
Sequence : 99
Group    : 00
Checksum : 91

fut091 packet received (9 bytes):
Raw packet: B8 24 B1 1A AE 15 6A AD D5 

Decoded:
Key      : B8
b1       : 21
ID       : 7853
Command  : 01
Argument : 05
Sequence : 98
Group    : 00
Checksum : 8A
```
### Setup information

ESP8266 Wemos D1 mini with nRF24L01 shield

#### Firmware version
1.10.3 (d1_mini)

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.3","ip_address":"192.168.0.40","reset_reason":"Power on","variant":"d1_mini","free_heap":17816,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":100,"http_repeat_factor":1,"auto_restart_period":360,"mqtt_server":"192.168.0.23","mqtt_username":"mqtt","mqtt_password":"xxxxxxx","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milightupdate/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":5,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Off","led_mode_packet_count":0,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"HIGH","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":15,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":5000,"rf24_channels":["LOW"],"device_ids":[854],"gateway_configs":[],"group_state_fields":["state","level","hue","saturation","mode","color_temp","bulb_mode"],"group_id_aliases":{}}
```

### Additional context

sidoh, 2019-11-02:
Unfortunately this is just how the remotes behave.  A "held" on/off button is interpreted by bulbs as "night mode."

You could try subscribing to the state topic rather than the updates topic, and inspecting the `status` field from there.  It'll be `"ON"`, even when in night mode.

RURon, 2019-11-03:
Ah okay, interesting ;-) Thank you for clearing this up.
If I understand things correctly, when I have the situation that my button press is not recognized directly and therefore the ESP only receives the night mode message, the state topic won't really help, as the desired button press message would not be visible to the ESP, either.

So what I came up with, is to assume that the user wants to change the state of a channel (on/off) if we receive a night mode message on a channel and the last change is longer ago than for example 5 seconds or so. This gives a chance to have the result that I want to trigger when using the remote.

I now see another bug, I will open in another ticket if I cannot find it already opened....

sidoh, 2019-11-03:
Ah, yeah, I guess night mode alone doesn't put the bulb in "on."

There's no "last changed" timestamp for bulb state, and it's probably not worth adding one just for this.

Are you able to respond to night_mode commands in addition to status=ON in your automation?

btw- night mode only happens when you're holding down the button.  If you press on "ON" multiple times, you'll get multiple ON commands.

RURon, 2019-11-04:
Yes, I'm using OpenHAB2 and in one single rule I work on everything that is coming from the fut091 remotes and handle all messages there instead of only triggering one action thru one message. So I can check on a "night_mode", when I last changed the "state" of that channel and thus can eventually assume that the user indeed wants to change the state of this channel (so that I can drop repeated night_mode messages).

Concerning the holding down vs. repeated short presses: I found out that when the radio range is becoming the issue, then the "holding down and waving the remote in the air" is easier than "waving in the air and repeatedly pressing the button" ;-)

That works quite well at the moment. Now I only hope that the MQTT update setting stays as it is :-) - the only work around (for me) would be to buffer the state message and look at changes in order to get the recent button press. For once I removed the auto restart...maybe that actually helps.

BTW: Thank you for this nice Gateway and greetings from Germany!

sidoh, 2019-11-16:
Sounds like you were able to find a solution.  Let me know if I'm misunderstanding and we can re-open.

closed 2019-11-16 by sidoh