sidoh/esp8266_milight_hub #548: MQTT update topic setting gets cleared at times

RURon, 2019-11-03:
### Describe the bug

I noticed that I'm loosing the MQTT update topic string at times. So far I cannot see a relationshio to anything, my only guess that maybe the auto restart that I enabled has to do with it. When this happens, that setting for the MQTT update topic field turns empty and the MQTT updating does not happen anymore. After rebooting the ESP, this is then still empty and I have to manually change the setting to make it work again 

### Setup information

ESP8266 Wemos D1 mini with nRF24L01 shield

#### Firmware version
1.10.3 (d1_mini)

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.10.3","ip_address":"192.168.0.40","reset_reason":"Power on","variant":"d1_mini","free_heap":17816,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":100,"http_repeat_factor":1,"auto_restart_period":360,"mqtt_server":"192.168.0.23","mqtt_username":"mqtt","mqtt_password":"xxxxxxx","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milightupdate/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":5,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Off","led_mode_packet_count":0,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"HIGH","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":15,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":5000,"rf24_channels":["LOW"],"device_ids":[854],"gateway_configs":[],"group_state_fields":["state","level","hue","saturation","mode","color_temp","bulb_mode"],"group_id_aliases":{}}
```

sidoh, 2019-11-03:
Maybe related to #523?  Unfortunately haven't been able to find/reproduce the memory leak.

RURon, 2019-11-04:
Too bad, happened again today. I now updated the firmware to 1.10.4 (d1_mini) - and see what happens. I might try older releases as well...maybe I can find one that works for me

RURon, 2019-11-07:
Interestingly, after I cleared the state topic, I so far have a steady situation as the update topic remains stable and that is currently what I need most...

RURon, 2019-11-16:
I think I found the cause for this: I am using OpenHab2 with the "binding" to connect to the ESP8266 Milight Hub. This binding seems to do the MQTT topic setting adjustments and simply clears the update topic and sets a state topic...which makes it unuseable for me unfortunately. Once the binding is removed, I have no issues anymore...

sidoh, 2019-11-16:
Glad you found the issue.  You could bring it up with the author on the OpenHAB forums.  I think this is the appropriate thread:

https://community.openhab.org/t/use-esp8266-to-control-unlimited-groups-of-milight-or-limitless-led-bulbs-in-openhab2/34906

closed 2019-11-16 by sidoh

JMacIV, 2019-11-16:
OH user here, this is the main thread for the binding - https://community.openhab.org/t/espmilighthub-new-binding-for-milight-limitlessled-and-easybulb/39720

sidoh, 2019-11-17:
Thanks for the pointer, @JMacIV!