sidoh/esp8266_milight_hub #550: Is there an easier way?

agoralive, 2019-11-03:
[image]

What I do is:

1. Receive a message from MQTT server
2. Converts into json
3. Moves to msg.payload
4. Determine how many clicks the Xiaomi Button got
5. Reads the file that contains latest state
6. Directs the flow to the opposite direction of the file contents
7. Sets the payload to on or off
8.1. Sends the mqtt to ESP8266 Milight Hub
8.2. Prepares for change the state within the log file
9. Saves the latest state into the log file, and is it ready for next "push of a button"

I have multiple ESP8266 Milight Hub's running, the TOGGLE funtion will not work here :(.

I would like to keep it simple, so no database solutions please....

sidoh, 2019-11-03:
What problem are you trying to solve?

agoralive, 2019-11-03:
Trying to find a "simpler" solution for the flow above

sidoh, 2019-11-03:
Right.  Can you please summarize what you're trying to accomplish with the flow?

agoralive, 2019-11-03:
This flow allows a Xiaomi switch to control the milight bulbs using mqtt. 

Only the toggle function can not be used due to multiple milight hubs running.

sidoh, 2019-11-03:
It seems like [context](https://nodered.org/docs/user-guide/context#context-scopes) is a pretty natural solution, no?  Not functionally different than reading from a file, just makes fewer assumptions about environment.

sidoh, 2019-11-07:
Gonna close since my understanding is that this isn't really an issue with espMH.  Happy to discus further, though.

closed 2019-11-07 by sidoh