sidoh/esp8266_milight_hub #551: Transition POST operation issue

kgoik, 2019-11-04:
Hi

I'm trying to use the transition endpoint to gradually change the brightness of my bulb. I've found a problem with either the code or docs:

In the docs there is a filed called device_type

{

    "field": "brightness",
    "start_value": 0,
    "end_value": 0,
    "duration": 0,
    "period": 0,
    "id": 0,
    "last_sent": 0,
    "bulb": 

    {
        "device_id": 1234,
        "group_id": 1,
        **"device_type"**: "rgb_cct"
    },
    "current_value": 0

}
Where the code is looking for following params:

void MiLightHttpServer::handleCreateTransition(RequestContext& request) {
  JsonObject body = request.getJsonBody().as<JsonObject>();

  if (! body.containsKey(**GroupStateFieldNames::DEVICE_ID**)
    || ! body.containsKey(**GroupStateFieldNames::GROUP_ID**)
    || ! body.containsKey(**F("remote_type"))**) {
    char buffer[200];
    sprintf_P(buffer, PSTR("Must specify required keys: device_id, group_id, remote_type"));

    request.response.setCode(400);
    request.response.json[F("error")] = buffer;
    return;
  }

I've tried to use different values both with device_type and remote_type but nothing works. 

### Steps to reproduce

invoke https://host/transitions POST method with 

### Expected behavior

### Setup information

#### Firmware version

#### Output of http://milight-hub.local/about

```json
"... /about output. put between the ```s"
```

#### Output of http://milight-hub.local/settings

```json
"... /settings output. put between the ```s"
```

### Additional context

sidoh, 2019-11-16:
Hello,

Oops, I thought I had fixed this, but apparently not.  I'll fix.

sidoh, 2019-11-17:
Fixed in [1.10.5](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.10.5).

sidoh, 2019-11-17:
(btw, the correct structure for `POST` is this):

```json
{
  "field": "brightness",
  "start_value": 0,
  "end_value": 255,
  "duration": 10,
  "device_id": 1234,
  "group_id": 1,
  "device_type": "rgb_cct"
}
```

Docs should reflect this now:

https://sidoh.github.io/esp8266_milight_hub/branches/1.10.5/

sidoh, 2019-11-17:
Thanks for reporting!

closed 2019-11-17 by sidoh