sidoh/esp8266_milight_hub #552: Support for passively listening on REST API requests

buenyamin-olgun, 2019-11-07:
Hello,
I'm right now realizing a homebridge plugin via REST API to mitigate #522 
Right now there's a lack of the status of the bulbs to keep them in sync with the remotes.

The GET request is only responding with ``
{
    "success": true
}
``. It would be helpful if the platform could store the last requests for each zone per gateway (maybe local, maybe in a mysql remote database on the pi where homebridge is hosted) and respond with the real properties. To keep things smooth this would be an opt-in feature so that stateful responses would need some extra parameters or sth like that.

Is this possible?

I know that there's the MQTT way for this but the REST API should also be stateful imho.

Thank you for your great work, keep it up!

Kind Regards

sidoh, 2019-11-07:
If you provide `?blockOnQueue=true`, it'll return the last known state.

https://sidoh.github.io/esp8266_milight_hub/branches/latest/#tag/Device-Control/paths/~1gateways~1{device-id}~1{remote-type}~1{group-id}/get

buenyamin-olgun, 2019-11-10:
Thanks for this, didn't know that this parameter stands for return of last known state.

closed 2019-11-10 by buenyamin-olgun