sidoh/esp8266_milight_hub #556: Bulbs dont react in any way

123vollhorst, 2019-11-15:
Hi there,
after spending the whole day figuring out why it doesnt work I hope you guys may find a direction to go.

Problem
None of my three bulbs react when trying to control them within the webUI though sniffer sees packages sent. With the Milight-App and via Openhab there isnt any controlling issue.

Setup
- (working) iBox2 
- (working) Milight bulbs (2x RGB-CCTs, 1x RGBW) 
- new NodeMCUv2  
- 3 new nRF24s

### What firmware version(s) have you tried? 
- The 1.0.14 ESP8266.bin

### Have you tried controlling the device with a physical remote?
- Dont have one yet, will arrive in the next days

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 
<details>
  <summary>settings</summary>

  |  
-- | --
admin_username | ""
admin_password | ""
ce_pin | 4
csn_pin | 15
reset_pin | 0
led_pin | -2
radio_interface_type | "nRF24"
packet_repeats | 50
http_repeat_factor | 1
auto_restart_period | 0
mqtt_server | ""
mqtt_username | ""
mqtt_password | ""
mqtt_topic_pattern | ""
mqtt_update_topic_pattern | ""
mqtt_state_topic_pattern | ""
mqtt_client_status_topic | ""
simple_mqtt_client_status | false
discovery_port | 48899
listen_repeats | 3
state_flush_interval | 10000
mqtt_state_rate_limit | 500
packet_repeat_throttle_sensitivity | 0
packet_repeat_throttle_threshold | 200
packet_repeat_minimum | 3
enable_automatic_mode_switching | false
led_mode_wifi_config | "Fast toggle"
led_mode_wifi_failed | "On"
led_mode_operating | "Slow blip"
led_mode_packet | "Flicker"
led_mode_packet_count | 3
hostname | "milight-hub"
rf24_power_level | "MAX"
rf24_listen_channel | "LOW"
wifi_static_ip | ""
wifi_static_ip_gateway | ""
wifi_static_ip_netmask | ""
packet_repeats_per_loop | 10
home_assistant_discovery_prefix | ""
wifi_mode | "n"
default_transition_period | 500
rf24_channels |  
0 | "LOW"
1 | "MID"
2 | "HIGH"
device_ids |  
0 | 4166
1 | 4167
gateway_configs |  
0 |  
0 | 4166
1 | 5987
2 | 6
1 |  
0 | 4166
1 | 8899
2 | 6
2 |  
0 | 4166
1 | 48899
2 | 6
group_state_fields |  
0 | "state"
1 | "brightness"
2 | "computed_color"
3 | "mode"
4 | "color_temp"
5 | "bulb_mode"
group_id_aliases |  
Test |  
0 | "rgb_cct"
1 | 4166
2 | 1
</details>
<details>
  <summary>about</summary>
{"firmware":"milight-hub","version":"1.10.4","ip_address":"192.168.1.18","reset_reason":"Power on","variant":"nodemcuv2","free_heap":16608,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
</details>

The Exclusion Path I gone (What I have done so far):
- I flashed the nRF24s via Linux and Windows
- I tried out all three nRF24s here
- I changed the jumper cables
- All Devices in one Room
- Played around with a variety of settings regarding the iBox; attached some screens
https://i.imgur.com/jnj56aV.png   
https://i.imgur.com/6vHS6aI.png

I have to admit that I  changed more than one "probable cause variable" while trying to get this working, but more in the later software phases (iBox config..). 

Thanks guys
Kelvin

sidoh, 2019-11-16:
Hello,

Have you worked through the [troubleshooting guide](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting)?

In particular, both bullet points here will narrow the space of possible problems:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting#make-sure-sends-work

123vollhorst, 2019-11-16:
> Have you worked through the troubleshooting guide?
Yes, as good as I can. Some thoughts here:

> Enable sniffing in the UI
With app and remote all times traffic/command were sent. When "creating" a fitting remote via device ID traffic is sent also by their ID. No obvious difference compared to a functional command via remote/app

> Hardware
What I ordered. I think I will get another ESP, maybe a WeMo mini. Dont know, shouldnt be something more expensive than it must be. 
https://www.amazon.de/gp/product/B06XJN417D
https://www.amazon.de/gp/product/B06Y1LZLLY

> Wiring
Also cheap cables but I tried two sets. Are there connections that could explain sth. like this? Then I will double check it.

> Interference
Well, I have a lot of tech (hardware and software) here within my little home cinema, but ESP and nRF are never more than 1m away from the next bulb.

Hope it helps, the hub is the perfect gateway for the lighting system I design here at home. Thanks for the solution and your support.

sidoh, 2019-11-16:
Sounds like you've maybe not tried the 2nd bullet in "making sure sends work" --

> Set up another ESP8266 with an nRF and flash esp8266_milight_hub on it as well. Enable sniffing on that second ESP while sending commands with the first.

This will allow you to verify packets are being sent.

Interference is mostly a problem when the nRF24 and the ESP8266 are really close to one another.  Other stuff >1' away shouldn't matter.

123vollhorst, 2019-11-17:
> This will allow you to verify packets are being sent.

Yes, couldnt untill now due to lack in proper hardware. Until I get my second ESP I try to figure out if I can use my RasPi instead of an ESP. If you have any suggestions how to use it with the nRF24 attached as a proper sniffer feel free for a hint, otherwise I have to wait (grr....) for the ESP.
Initial RP nRF24 config
`pi@picroft:~/rf24libs/RF24/examples_linux $ sudo ./gettingstarted_call_response
RPi/RF24/examples/gettingstarted_call_response
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x65646f4e32 0x65646f4e31
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x65646f4e32
RX_PW_P0-6	 = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x3f 0x06
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
`

sidoh, 2019-11-17:
Gotcha.  No experience with nRF+RPi personally, so won't be of much help.

123vollhorst, 2019-11-19:
> Set up another ESP8266 with an nRF and flash esp8266_milight_hub on it as well. Enable sniffing on that second ESP while sending commands with the first.

Done. Got another NodeMCUV2. Both are transmitting correctly. Then with the remote:

Working Remote

```
rgb_cct packet received (9 bytes):
Raw packet: BA F0 CC A2 1B A7 43 DC A1 

Decoded:
Key      : BA
b1       : 20
ID       : 15E4
Command  : 02
Argument : 4B
Sequence : 2E
Group    : 03
Checksum : 95

rgb_cct packet received (9 bytes):
Raw packet: CF 1D 05 30 D6 79 0F E9 FB 

Decoded:
Key      : CF
b1       : 20
ID       : 15E4
Command  : 02
Argument : 49
Sequence : 2C
Group    : 03
Checksum : 76
```

The not working Hub
```
rgb_cct packet received (9 bytes):
Raw packet: 00 DB CE BF 66 D0 B5 65 D0 

Decoded:
Key      : 00
b1       : 20
ID       : 15E4
Command  : 01
Argument : 00
Sequence : 07
Group    : 00
Checksum : D9

rgb_cct packet received (9 bytes):
Raw packet: 00 DB CE BF 66 CD B4 65 CC 

Decoded:
Key      : 00
b1       : 20
ID       : 15E4
Command  : 01
Argument : 05
Sequence : 06
Group    : 00
Checksum : DD
```
Maybe I did something wrong within the setup process. I know computers pretty well, but here I am fighting error messages the most time. A bit out of my skill zone...

123vollhorst, 2019-11-23:
**Got it!**
Ordered some other [NRF24L01](https://www.amazon.de/gp/product/B07P95X6HM/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) and it works out of the box. 
Will investigate more later, but it seems that buying cheap parts in general doesnt seem to be the best idea here. I hope this insight helps others fighting with similar problems.
Cheers
Kelvin

closed 2019-11-23 by 123vollhorst