sidoh/esp8266_milight_hub #557: Stops sending commands unless settings are saved

dmacias72, 2019-11-16:
I was using a Adafruit Huzzah initially and switch to a nodemcev2 thinking the flash was corrupt. I only have RGBW protocal 5 bulbs.  I am currenty rolled back to 1.7.0 and have been fine for weeks. I started noticing this after 1.7.0. It would happen intermittently (once every few days, maybe even a week) where bulbs wouldn't respond, even from the webpage. I figured out that if I just went to settings and clicked save, they would start working again. This has gotten progressively worse and on later versions. bulbs become unresponsive in sometimes 5 min - 1 hour. I have hooked up to serial connection and waited till they were not working and tried sending commands but nothing showed up.

sidoh, 2019-11-16:
I'm happy to help troubleshoot, but there's not much actionable detail here.  I need to be able to reproduce the issue, or at least have a starting point.

Some pointers on things you can do:

* Paste the serial logs?
* Specify the requested information in the [issue template](https://github.com/sidoh/esp8266_milight_hub/issues/new?assignees=&labels=bug&template=bug_report.md&title=), especially the output of `/settings` and `/about`.
* What integration are you using?  REST?  MQTT?  UDP?

Saving settings re-initializes everything, including the connection to MQTT.  MQTT is [does](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MQTT/MqttClient.cpp#L122-L132) reconnect by itself, though.

oakbrad, 2019-11-19:
I believe I am having the same problem after upgrading firmware recently. I am sorry I did not note what version I was running before I upgraded - at least 1.10.0 as I've been using transitions.  

My hub has soldered connections and has had no problems for over a year so I don't believe it is because of range/interference/hardware/etc. I am using a Wemos D1 Mini + NRF antenna.

I am not seeing anything of note in my serial monitor.
```
11:35:15.437 -> ding parameter
11:35:15.437 -> 
11:35:15.437 -> *WM: netmask
11:35:15.437 -> 
11:35:15.437 -> *WM: Adding parameter
11:35:15.437 -> 
11:35:15.437 -> *WM: gatewa
```

/settings
```{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":5,"auto_restart_period":0,"mqtt_server":"192.168.0.9:1883","mqtt_username":"milight","mqtt_password":"XXXXXX","mqtt_topic_pattern":"lights/commands/:device_alias","mqtt_update_topic_pattern":"lights/updates/:device_alias","mqtt_state_topic_pattern":"lights/states/:device_alias","mqtt_client_status_topic":"milight_hub/status","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Off","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"HIGH","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"g","default_transition_period":500,"rf24_channels":["MID","HIGH"],"device_ids":[8192,23365,24576,26214,32768,47872,65280],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","mode","bulb_mode","computed_color","effect"],"group_id_aliases":{"bed_strip":["rgbw",47872,3],"bedroom":["rgbw",26214,1],"dining_floor_lamp":["rgbw",32768,1],"entry":["rgbw",8192,1],"floor_lamp":["rgbw",32768,3],"fridge":["rgbw",24576,1],"globes":["rgbw",32768,4],"middle_pendant":["rgbw",8192,2],"office_pendant":["rgbw",8192,3],"plant_accent":["rgbw",65280,1],"reading_lamp":["rgbw",24576,2],"stereo_strip":["rgbw",47872,2],"stove":["rgbw",32768,2]}}```

/about
```
{"firmware":"milight-hub","version":"1.10.4","ip_address":"192.168.0.102","reset_reason":"External System","variant":"d1_mini","free_heap":15400,"arduino_version":"2_4_2","queue_stats":{"length":13,"dropped_packets":0}}
```

I am using MQTT (Mosquitto) and I do see it timing out with a socket error in my logs:
```
1574191351: Socket error on client milight-hub-3408445, disconnecting.
1574191354: New connection from 192.168.0.102 on port 1883.
[INFO] found milight on local database
1574191355: New client connected from 192.168.0.102 as milight-hub-3408445 (p2, c1, k15, u'milight').
1574191377: Socket error on client milight-hub-3408445, disconnecting.
1574191380: New connection from 192.168.0.102 on port 1883.
1574191380: New client connected from 192.168.0.102 as milight-hub-3408445 (p2, c1, k15, u'milight').
1574191546: Socket error on client milight-hub-3408445, disconnecting.
```

Let me know if there is more information I can provide.

sidoh, 2019-11-30:
Thanks for the details, @oakbrad.  There seem to be 25 seconds between reconnect attempts given your logs.  Is that happening consistently?

dmacias72, 2019-11-30:
I believe my problem was a power issue or nrf issue. I swapped out nrf modules and also powered it separately with a spare 5v nrf socket adapter. I upgraded to latest version and have been running all week without problems. Still not sure why 1.7.0 worked but later version didn't. But I believe the nrf was becoming unstable.

oakbrad, 2019-11-30:
I think mine is hardware related as well - I set up a spare hub and imported my config and so far so good over the last few days. It definitely started having issues immediately after upgrading firmware, so I'm not sure why that would effect a previously working hub.

dmacias72, 2019-11-30:
I would say this issue is closed

closed 2019-11-30 by dmacias72