sidoh/esp8266_milight_hub #558: Device discovery v6 protocol

rremigius, 2019-11-18:
Hi, first of all great piece of software! So far ahead on any other attempts to control Milights programmatically.

I'm trying to control my Milight RGB+CCT bulbs with the official (and other) apps through the es8266 milight hub, but none of the apps are able to discover the hub as a controller.

First, some info: 
- I've read the [troubleshooting](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting#make-sure-sends-work) guide.
- I'm using a nodemcu board.
- Firmware version: 1.10.4-2-g62c538e
- Built the firmware from latest source (last week)
- Trying to access a FUT105 (also tried with FUT014)
- Radio type RGB+CCT
- [/about](https://github.com/sidoh/esp8266_milight_hub/files/3860465/about.txt)
- [/settings](https://github.com/sidoh/esp8266_milight_hub/files/3860466/settings.txt)

## What works

- Access to the milight-hub.local web page.
- Pairing the bulbs and controlling them from the web page
- Sniffing remote controller packets
- Adding a UDP device for 0x1, UDP port 5987, protocol version 6 in settings
- Controlling the bulbs from the unofficial [Milight Controller](https://play.google.com/store/apps/details?id=com.epocapp.milightcontroller&hl=en) app, by setting the controller's IP address and port manually. *
- When I add a UDP device using port 8899 and version 5, the official [Mi-Light Remote](https://play.google.com/store/apps/details?id=mi_light.mi_light&hl=en) app can discover the controller - however, obviously, the bulb will not respond because the protocol does not support RGB+CCT.

\* This observation makes me think there is only something wrong with the discovery and it will work once properly connected.

## What doesn't work
- Using the official [Mi-Light Remote](https://play.google.com/store/apps/details?id=mi_light.mi_light&hl=en) app, the hub will not be discovered if using protocol v6, port 5987.
- The official [Milight 3.0](https://play.google.com/store/apps/details?id=com.irainxun.wifilight&hl=en) app attempts some configuration with wifi ssid and password, but also without any success.

I hope this information gives enough insight into the issue to identify the problem. I hope you could give me some pointers on how to propery set the system up to connect properly with the official MiLight app(s).

sidoh, 2019-11-30:
Hello, thanks for reporting.

The v6 protocol uses a different discovery sequence which is enabled with this setting:

[image]

I can't guarantee that this will work, but it is at least necessary for the hub to be detected.

If that doesn't work, something has probably changed.  The packets in the UDP v6 protocol have changed in content or structure at least twice since I started this project.  It's undocumented, so each time involves fairly painful reverse-engineering.

I'd be happy to take a look at a PR, but not planning on doing this again myself because:

* The official milight app isn't terribly popular
* It'll probably break again in the future

rremigius, 2019-12-09:
Hi, thanks for the reply.

The Discovery Port is already set to 48899, so unfortunately that doesn't seem to work.
In the meantime I built my own web app for it (which was the final plan anyway), so I won't need the official app anymore :)

Thanks for the great library!

sidoh, 2019-12-15:
Gotcha, thanks for following up.

(btw, thanks very much for the including all of the details you did in the initial report.  very thorough :).

closed 2019-12-15 by sidoh