sidoh/esp8266_milight_hub #562: Not conecting to RGBW Led strips

hpuk2000, 2019-12-06:
I love this project and was thinking it was the answer to finaly integrating my 9 Milight RGBW strip lights that I use as my main kitchen lighting, into my extensive OpenHab installation running on a Raspberry Pi 3B+

I have put together 2 EspHubs using NodeMCU V2's, one with an nRF24L01 and one with an nRF24L01+PA+LNA.

Both flashed with the latest FW, 1.10.5.

Web interface works perfectly and I can sniff both hubs and my 2 remotes.
MQTT works perfectly too and also from the remotes. I'm using MQTT.fx to monitor and mosquitto run on my OpenHab Pi.

The only problem is the web interface will not control the RGBW strips, and the codes being sent out on any remote Type do not match the codes from my physical remotes.

I guess this is not so much of a problem, more of an enhancement/update rquest.

My 9 Led controllers are FUT038's and my remotes are both FUT096.

If there was some option in the web interface to create my own packet from sniffing the remote I would probably get this working.
Failing that is there any chance the FW could be updated to emulate the FUT096 remote?

Thank you.
Chris.

hpuk2000, 2019-12-13:
Update:-

This seems to be working perfectly now. 
Only changes I made were to swap the 1 nRF24L01+PA+LNA to a standard nRF24L01. Both Mcu's are  sitting within 3 feet of each other so don't know if there was interferrence.
I powered on my iBox again as I'd switched it off for troubleshooting.
I also started using the device ID of my second remote but both ID's now work.
I did order a FUT014 bulb and a FUT89 remote and they worked fine with my setup straight away but now don't for some reason but I don't want to control that bulb anyway.

So I'm not absolutley what fixed it but I'm very pleased.
Thank you @sidoh  for this great project.

Chris

sidoh, 2019-12-15:
Glad you got it working.

I'm always a little bit amazed at how flakey nRF24L01s seem.

closed 2019-12-15 by sidoh