sidoh/esp8266_milight_hub #563: Control milight with sound sensor

Catskan, 2019-12-06:
Hi guys,

My diy gateway works fine now ! But now, I want to use a sound sensor but I don't know how to make this and if it's possible ?

I search on the web to 1h but I haven't find any result for it ...

Thanks you in advance.

Regards

poudenes, 2019-12-08:
There was a tool to do this. 

https://www.epocapp.eu/milight/

But since september 2018 there are no updates...

I never test this.

Catskan, 2019-12-09:
Hi,

Yes, I found this but is only with a software. Is it possible to use a hardware sensor with a arduino or directly with the Nocemcu3 ?

Thanks

sidoh, 2019-12-15:
Hi @Catskan,

I'm sure this is possible in theory, but not without significant code modifications.

Not something I would plan to put on the roadmap anytime soon, so going to close.

closed 2019-12-15 by sidoh