sidoh/esp8266_milight_hub #564: device name

poudenes, 2019-12-09:
1.10.5 (nodemcuv2)

When add device name to groups (1,2,3,4,0) in beginning it goes well. But when you add more at certain moment it seems you add the names correct, you see them in the list but the whole config is empty.

My Hub is in running config so coat test to much. I can restore the config file without the device names then everything is working again.

poudenes, 2019-12-09:
Here the JSON how it must look like (completed in a text editor)

```
{
    "admin_username":"",
    "admin_password":"",
    "ce_pin":16,
    "csn_pin":15,
    "reset_pin":0,
    "led_pin":-2,
    "radio_interface_type":"nRF24",
    "packet_repeats":50,
    "http_repeat_factor":1,
    "auto_restart_period":0,
    "mqtt_server":"192.x.x.x:1883",
    "mqtt_username":"USERNAME",
    "mqtt_password":"PASSWORD",
    "mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id",
    "mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id",
    "mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id",
    "mqtt_client_status_topic":"milight/client_status",
    "simple_mqtt_client_status":true,
    "discovery_port":48899,
    "listen_repeats":0,
    "state_flush_interval":100,
    "mqtt_state_rate_limit":500,
    "packet_repeat_throttle_sensitivity":50,
    "packet_repeat_throttle_threshold":500,
    "packet_repeat_minimum":50,
    "enable_automatic_mode_switching":false,
    "led_mode_wifi_config":"Fast toggle",
    "led_mode_wifi_failed":"Fast blip",
    "led_mode_operating":"Slow toggle",
    "led_mode_packet":"Fast blip",
    "led_mode_packet_count":4,
    "hostname":"MilightHub",
    "rf24_power_level":"MAX",
    "rf24_listen_channel":"LOW",
    "wifi_static_ip":"192.68.100.70",
    "wifi_static_ip_gateway":"192.168.100.1",
    "wifi_static_ip_netmask":"255.255.255.0",
    "packet_repeats_per_loop":50,
    "home_assistant_discovery_prefix":"",
    "wifi_mode":"n",
    "default_transition_period":500,
    "rf24_channels":[
        "LOW",
        "MID",
        "HIGH"
    ],
    "device_ids":[
        43681,
        43682,
        43683,
        43684,
        43690,
        43691,
        43692,
        43693,
        43694,
        43695,
        43706,
        43707
    ],
    "gateway_configs":[

    ],
    "group_state_fields":[
        "state",
        "status",
        "brightness",
        "level",
        "hue",
        "saturation",
        "color",
        "mode",
        "kelvin",
        "color_temp",
        "bulb_mode",
        "computed_color",
        "effect",
        "device_id",
        "group_id",
        "device_type"
    ],
    "group_id_aliases":{
        "balcony":[
            "rgb_cct",
            43684,
            1
        ],
        "bathroom_all":[
            "rgb_cct",
            43693,
            0
        ],
        "bathroom_1":[
            "rgb_cct",
            43693,
            1
        ],
        "bathroom_2":[
            "rgb_cct",
            43693,
            2
        ],
        "bathroom_3":[
            "rgb_cct",
            43693,
            3
        ],
        "bathroom_4":[
            "rgb_cct",
            43693,
            4
        ],
        "bedroom_1":[
            "rgb_cct",
            43690,
            1
        ],
        "bedroom_2":[
            "rgb_cct",
            43690,
            2
        ],
        "bedroom_3":[
            "rgb_cct",
            43690,
            3
        ],
        "bedroom_4":[
            "rgb_cct",
            43690,
            4
        ],
        "bedroom_5":[
            "rgb_cct",
            43691,
            1
        ],
        "bedroom_6":[
            "rgb_cct",
            43691,
            2
        ],
        "bedroom_all":[
            "rgb_cct",
            43706,
            1
        ],
        "dressoir_all":[
            "rgb_cct",
            43695,
            0
        ],
        "dressoir_1":[
            "rgb_cct",
            43695,
            1
        ],
        "dressoir_2":[
            "rgb_cct",
            43695,
            2
        ],
        "dressoir_3":[
            "rgb_cct",
            43695,
            3
        ],

        "hallway":[
            "rgb_cct",
            43683,
            1
        ],
        "kitchen_all":[
            "rgb_cct",
            43682,
            0
        ],
        "kitchen_1":[
            "rgb_cct",
            43682,
            1
        ],
        "kitchen_2":[
            "rgb_cct",
            43682,
            2
        ],
        "kitchen_3":[
            "rgb_cct",
            43682,
            3
        ],
        "livingroom_all":[
            "rgb_cct",
            43692,
            0
        ],
        "livingroom_wall_right":[
            "rgb_cct",
            43692,
            1
        ],
        "livingroom_wall_left":[
            "rgb_cct",
            43692,
            2
        ],
        "livingroom_window_left":[
            "rgb_cct",
            43692,
            3
        ],
        "livingroom_window_right":[
            "rgb_cct",
            43692,
            4
        ],
        "livingroom_wall":[
            "rgb_cct",
            43707,
            1
        ],
        "livingroom_window":[
            "rgb_cct",
            43707,
            2
        ],
        "sink_all":[
            "rgb_cct",
            43681,
            0
        ],
        "sink_1":[
            "rgb_cct",
            43681,
            1
        ],
        "sink_2":[
            "rgb_cct",
            43681,
            2
        ],
        "sink_3":[
            "rgb_cct",
            43681,
            3
        ],
        "toilet_all":[
            "rgb_cct",
            43694,
            0
        ],
        "toilet_1":[
            "rgb_cct",
            43694,
            1
        ],
        "toilet_2":[
            "rgb_cct",
            43694,
            2
        ],
        "toilet_3":[
            "rgb_cct",
            43694,
            3
        ]
    }
}
```

sidoh, 2019-12-15:
Unfortunately this is an unstated limitation of any of the expandable configurations with HASS.  All configs are stored as a giant JSON blob, and the buffer used to read settings has a fixed max size of 4KB.

It should definitely not be the case that your settings get wiped if you try to go beyond this buffer size, but it does not surprise me that it behaves this way.

The real fix is to split some of this stuff out into separate routes (e.g., `/aliases`), but that's a bigger change than I have time for right now.

As a workaround, you can increase the size of this buffer by compiling with `MILIGHT_HUB_SETTINGS_BUFFER_SIZE=8196` (or whatever value makes sense).

poudenes, 2019-12-15:
Thanks for the info. I will only give the group a name Then its ok for me also :)

closed 2019-12-15 by poudenes