sidoh/esp8266_milight_hub #565: Home Assistant discovery issue

joshdinsdale, 2019-12-10:
hi Guys

I have just migrated to home assistant from Domoticz and am having some issues with my espMH.

I have followed the guide for home assistant autodiscovery (i know auto discovery works in HA as i have another mechanism using it), however my device isnt showing in HA correctly. I get a device, but it doesnt expose any entities properly. I can add it as a switch, but it doesnt show up for automations etc.
I dont seem to be seeing the correct MQTT data when i tune into it using a MQTT client, but im not sure why this is.

My espMH config is below:

`{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":1438,"mqtt_server":"192.168.7.19:1883","mqtt_username":"?????","mqtt_password":"????","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"g","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[9761,10496],"gateway_configs":[[10496,5987,5]],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Master Bedroom Lamp":["cct",10496,1]}}`

Any ideas?

J

joshdinsdale, 2019-12-10:
OK, so if i tune into the correct topic I can see the updates which seem correct:

`{
    "schema": "json",
    "name": "Master Bedroom Lamp",
    "command_topic": "milight/0x2900/cct/1",
    "state_topic": "milight/states/0x2900/cct/1",
    "device": {
        "manufacturer": "esp8266_milight_hub",
        "sw_version": "1.10.5",
        "identifiers": [
            1770164,
            10496,
            "cct",
            1
        ]
    },
    "availability_topic": "milight/client_status",
    "payload_available": "connected",
    "payload_not_available": "disconnected",
    "brightness": true,
    "effect": true,
    "effect_list": [
        "night_mode"
    ],
    "color_temp": true
}`

But it doesnt show in the entities list, just devices.

sidoh, 2019-12-15:
This looks like what I'd expect given your configuration.  Only named devices are registered with HASS discovery, and looks like you've only named one of your devices:

```json
{
  "group_id_aliases": {
    "Master Bedroom Lamp": [
      "cct",
      10496,
      1
    ]
  }
}
```

name the other one and it should show up.

sidoh, 2019-12-15:
(btw, only edited your comments to change the inline code formatting to block code formatting)

joshdinsdale, 2019-12-15:
Hi Chris

Thanks for the reply. I ended up flattening my HA setup and starting again, i think it had some database corruption or something as i was having issues with other devices and autodiscovery.

AAAANYWAY... 
I now have an entry for my device on channel 1 in the devices menu in Configuration, however it doesnt seem to expose any entities under the entities menu. I know there is one as it gets added to my all lights group automatically and is added to the lovelace screen. I initially tried making an automation for it (turn light on at x time), however when i select it as a device it doesnt expose any actions. However if I manually create the automation and use a turn on/off action the automation runs as expected.

I'm quite new to HA so im possibly a little confused about how it works, I also understand its under constant development, so I dont know if what I am experiencing is an issue with the espMQ integration/mqtt integration or my HASSIO/HA setup itself.

Feel free to bounce me off to the home assistant forums if necessary! :)

Josh

caffeineflo, 2020-01-11:
I have the same issue, the problem is that the MQTT HA autodiscovery message isn't picked up by HA correctly.

The device is registered, but since the metadata for the MQTT integration is malformed for HA, it doesn't show up under entities or under the integration

yfrans, 2020-01-17:
I also have the same issue.
Topic: `homeassistant/light/milight_hub_14313876/rgb_cct_0x1_1/config`
Message:
`
{"schema":"json","name":"Outdoor1","command_topic":"milight/0x1/rgb_cct/1","state_topic":"milight/states/0x1/rgb_cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"1.10.5","identifiers":[14313876,1,"rgb_cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode","white_mode","0","1","2","3","4","5","6","7","8"],"rgb":true,"color_temp":true}
`

avolmensky, 2020-01-31:
I also have this problem. I would like to assign a light to an area, but only the hub shows up under the MQTT integrations