sidoh/esp8266_milight_hub #566: Effect speed in Home assistant

strumf666, 2019-12-11:
I don't see the option to change this, is it not implemented, missing from light control or something else? It does work from milight hub.

sidoh, 2019-12-15:
espMH integrates with HomeAssistant using the [`light.mqtt`](https://www.home-assistant.io/integrations/light.mqtt/) component.  I don't believe it supports effect speed.

You can control effect speed using the [REST API](https://sidoh.github.io/esp8266_milight_hub/branches/latest/#tag/Device-Control-by-Alias/paths/~1gateways~1{device-alias}/put) or through raw MQTT packets.

Without `light.mqtt` supporting this or building a custom HASS component, there's no way to directly control this in HASS.

123vollhorst, 2019-12-17:
Sorry for entering the thread, but I may have a pretty similar concern. I am experimenting with light transitions in HA. ATM the RGBCCT lights are integrated via MQTT Autodiscovery and I wonder if there may be a practical/theoretical limit regarding the espMH + RGBCCT lights transition period. I just set it to 10 ms, but it seems more like 1 second steps what brings me to the hypothesis that it is kind of overridden by HA.
Are there any other controllable factors that have an influence here?
Thanks
Kelvin

strumf666, 2019-12-22:
> espMH integrates with HomeAssistant using the [`light.mqtt`](https://www.home-assistant.io/integrations/light.mqtt/) component. I don't believe it supports effect speed.
>
> You can control effect [...]
Ok, thanks for the explanation.

sidoh, 2019-12-31:
Great, gonna close the issue.

@123vollhorst, I think HomeAssistant [measures transitions in seconds](https://www.home-assistant.io/integrations/light/).  It does not seem like decimal values are supported.

espMH uses milliseconds under the hood, but you're limited by the packet send rate.  This depends on the settings you're using, but is usually around 200-300ms/packet (meaning any transition shorter than that is meaningless).

closed 2019-12-31 by sidoh