sidoh/esp8266_milight_hub #569: Limit on Device Names?

mcwussow, 2019-12-17:
Is there a limit on the number of device names that can be entered?

It seems that after I add a number of them I get this error:

remoteTypeFromString: ERROR - tried to fetch remote config for type: null
remoteTypeToString: ERROR - tried to fetch remote config name for unknown type: 255

I have successfully entered 25 device names.  I need to enter many more than this.

Please let me know!  Thanks!

poudenes, 2019-12-17:
See my post

https://github.com/sidoh/esp8266_milight_hub/issues/564

same question with answer below

sidoh, 2019-12-31:
Thanks @poudenes.  

Unfortunately you're limited by the size of the JSON buffer used to parse the settings blob right now (which is 2KiB).

I'll open an issue with a clear outline of the technical solution, but it's not something I'll have the time to address myself for quite a while.

closed 2019-12-31 by sidoh

poudenes, 2020-01-01:
Will it help when im shorter the names?
```
livingroom_1 > lr_1
livingroom_2 > lr_2
livingroom_3 > lr_3
livingroom_4 > lr_4
livingroom_5 > lr_5
livingroom_6 > lr_6
```
then I safe character places? So I can give every bulb a name packed in same 2KiB space :)