sidoh/esp8266_milight_hub #570: Android App / connect with phone

zierbeek, 2019-12-17:
Hi all, 

just flashed  this wonderful thing to the nodemcu v3, and was able to connect and pair everything. 

Can I get to it with an app or webinterface?

sidoh, 2019-12-31:
You mean other than the [builtin UI](https://github.com/sidoh/esp8266_milight_hub#use-it)?  There are probably android apps that it works with out of the box.  [Bilight](https://apkpure.com/bilight/com.jaspergoes.bilight) probably does the trick.

zierbeek, 2019-12-31:
Hi Chris, the problem is i Cant Acces is from my phone... I configure the wifi from my phone and after that, i can only acces from my computer . The lights work then.

Happy 2020!

sidoh, 2020-01-01:
I see.  Why are you not able to access it from your phone?  If it's on the same wifi network, should be possible (although the mDNS pointer http://milight-hub.local won't work on Android).

Happy New Year!

zierbeek, 2020-01-01:
It is on the same wifi-network. I thought it could have been this problem : I have a router from the isp and an old access point which is converted to a repeater, it broadcasts the same network and ssid creating a 'mesh'. I disabled this one and tried again, without luck..

quadragin, 2020-01-04:
Can you ping the esp8266_milight_hub IP address from your phone?

zierbeek, 2020-01-04:
Sorry, how do I do that? :)

quadragin, 2020-01-04:
Install some sort of networking tools on your phone

sidoh, 2020-01-04:
Are you trying to access it via the IP, or some DNS pointer?

zierbeek, 2020-01-04:
Directly through 192.168.1.4

jellewie, 2020-01-07:
When I connect the ESP to the PC and open a serial com (just use Arduino serial monitor) and set it to a baudrate of 9600 the ESP will spit out it's IP (its somewhat like a network name if you dont know what a IP is)
`*WM: 192.168.1.127`
Make sure the ESP and mobile are in the same network, sidenote the router must allow crosstalk between devices, this is most of the times the default.
Go with your phone to the IP you will see somthing like [this](https://i.imgur.com/x3r25Vb.jpg) if it worked. you can also select 'Desktop site' to get [this](https://i.imgur.com/ocfqs4E.jpg).

**Steps (again)**
1. Open serial to ESP at 9600
2. restart ESP and get the IP from the serial monitor
3. make sure ESP and mobile are on the same network (and the router allows crosstalk between devices)
4. Connect to the IP  

**Summary**
- mDNS pointer http://milight-hub.local doesn't work (as mentioned before)
- connect though IP (and changing settings) do work on Android* (verified)

*Checked on Motorola, Wileyfox, and Google Pixel C, so seems to be globally android

zierbeek, 2020-01-07:
Yes, able to connect, wrong ip... sorry guys!

but next to that, is there an app?

sidoh, 2020-01-07:
Not aware of another android app that works directly with milight.

(and thanks for pitching in, @jellewie!)

closed 2020-01-07 by sidoh