sidoh/esp8266_milight_hub #572: Discovery Light Home Assistant

poudenes, 2019-12-19:
Hi All,

When you give devices a device name then only those are discovered by Home Assistant.
I have several lights given a name, I cant give all a name because of amount of space for the config file.

All devices with a nice name are discovered all other not come up. I tried to turn them on and off but nothing

Firmware: 1.10.5 (nodemcuv2)
HA Version: 0.102.3

`{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"IP NUMBER:1883","mqtt_username":"USER","mqtt_password":"PASSWORD","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":100,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":50,"packet_repeat_throttle_threshold":500,"packet_repeat_minimum":50,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"Fast blip","led_mode_operating":"Slow toggle","led_mode_packet":"Fast blip","led_mode_packet_count":4,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"IP NUMBER 70","wifi_static_ip_gateway":"IP NUMBER .1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":50,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"],"group_id_aliases":{"balcony":["rgb_cct",43684,1],"bathroom_all":["rgb_cct",43693,0],"bedroom_1":["rgb_cct",43690,1],"bedroom_5":["rgb_cct",43691,1],"bedroom_all":["rgb_cct",43706,1],"dressoir_all":["rgb_cct",43695,0],"hallway":["rgb_cct",43683,1],"kitchen_all":["rgb_cct",43682,0],"livingroom_all":["rgb_cct",43692,0],"livingroom_wall":["rgb_cct",43707,1],"livingroom_window":["rgb_cct",43707,2],"sink_all":["rgb_cct",43681,0],"toilet_all":["rgb_cct",43694,0]}}`

sidoh, 2019-12-31:
This is currently just how discovery works -- only named devices are submitted to the discovery topic.  I could be convinced it should work differently, but I think this makes sense.

poudenes, 2022-10-24:
Is there any way to use Device ID as name? When name is empty? Otherwise use name. Or is memory of esp not enough to create this code?

closed 2023-04-13 by poudenes