sidoh/esp8266_milight_hub #575: Cloning a FUT089	handle only 4 zone

l1q1d, 2019-12-30:
The FUT098 controls 8 zone and sniffing shows correctly group 5,6,7,8 but with the current interface we are unable to use the other groups even if we clone the same id.
Is there a define to edit to increase from 4 to 8 groups?

The work around would be pairing the 4+ with another id but in this way the hub will be out of sync.

sidoh, 2019-12-31:
Unfortunately there's not enough information here for me to be of much help.  Some information that would help:

* What bulbs/devices are you attempting to pair?
* Do you have an official milight remote that the bulbs are already paired with?  If so, can you please post some packets from it (put the hub in listen mode and press buttons on the remote)

l1q1d, 2020-01-01:
Hi Sidoh I cloned a FUT098 paired with a LS2:
This is the log:

GROUP 8 ON (nothing connected)

Traffic sniffed (sent and received)
fut089 packet received (9 bytes):
Raw packet: 8B 64 88 EA 1F F6 EC 2C 59 

Decoded:
Key      : 8B
b1       : 25
ID       : 5A92
Command  : 01
Argument : 08
Sequence : 87
Group    : 08
Checksum : D8

GROUP 7 ON (nothing connected)

fut089 packet received (9 bytes):
Raw packet: C4 14 DB E5 9A 07 70 9C D9 

Decoded:
Key      : C4
b1       : 25
ID       : 5A92
Command  : 01
Argument : 07
Sequence : 86
Group    : 07
Checksum : 92

GROUP 6 ON (nothing connected)

fut089 packet received (9 bytes):
Raw packet: 4C 8C 63 5D 12 7E EB 13 C6 

Decoded:
Key      : 4C
b1       : 25
ID       : 5A92
Command  : 01
Argument : 06
Sequence : 85
Group    : 06
Checksum : 07

GROUP 5 change color

fut089 packet received (9 bytes):
Raw packet: AD 05 62 B0 C5 B9 1F CA CA 

Decoded:
Key      : AD
b1       : 25
ID       : 5A92
Command  : 01
Argument : 14
Sequence : 84
Group    : 05
Checksum : 74
fut089 packet received (9 bytes):
Raw packet: 3A 6D 09 78 9B 17 70 56 40 

Decoded:
Key      : 3A
b1       : 25
ID       : 5A92
Command  : 02
Argument : 5B
Sequence : 83
Group    : 05
Checksum : 74
fut089 packet received (9 bytes):
Raw packet: 43 AC 40 12 6A E5 39 77 6B 

Decoded:
Key      : 43
b1       : 25
ID       : 5A92
Command  : 02
Argument : 61
Sequence : 82
Group    : 05
Checksum : 72
fut089 packet received (9 bytes):
Raw packet: 18 C0 2F 39 0B 4B 25 0A 99 

Decoded:
Key      : 18
b1       : 25
ID       : 5A92
Command  : 02
Argument : 6F
Sequence : 7F
Group    : 05
Checksum : 66
fut089 packet received (9 bytes):
Raw packet: 2A 7D F9 68 6B BB A8 26 5C 

Decoded:
Key      : 2A
b1       : 25
ID       : 5A92
Command  : 02
Argument : 87
Sequence : 7B
Group    : 05
Checksum : 68
fut089 packet received (9 bytes):
Raw packet: 33 BC 30 02 3A A6 7E 47 91 

Decoded:
Key      : 33
b1       : 25
ID       : 5A92
Command  : 02
Argument : 90
Sequence : 79
Group    : 05
Checksum : 68
fut089 packet received (9 bytes):
Raw packet: 2C AC 43 3D EF F1 3E F6 8C 

Decoded:
Key      : 2C
b1       : 25
ID       : 5A92
Command  : 02
Argument : 95
Sequence : 78
Group    : 05
Checksum : 69
fut089 packet received (9 bytes):
Raw packet: 75 3D 2A 08 3C 89 25 42 6F 

Decoded:
Key      : 75
b1       : 25
ID       : 5A92
Command  : 02
Argument : 9C
Sequence : 76
Group    : 05
Checksum : 67
fut089 packet received (9 bytes):
Raw packet: 92 15 61 40 43 7B C8 FE 7C 

Decoded:
Key      : 92
b1       : 25
ID       : 5A92
Command  : 02
Argument : AF
Sequence : 73
Group    : 05
Checksum : 60
fut089 packet received (9 bytes):
Raw packet: 6F 80 6C BE F6 8F B0 0B AC 

Decoded:
Key      : 6F
b1       : 25
ID       : 5A92
Command  : 02
Argument : BF
Sequence : 6F
Group    : 05
Checksum : 49
fut089 packet received (9 bytes):
Raw packet: B4 24 CB D5 A7 53 98 AE 20 

Decoded:
Key      : B4
b1       : 25
ID       : 5A92
Command  : 02
Argument : C3
Sequence : 6E
Group    : 05
Checksum : 45
fut089 packet received (9 bytes):
Raw packet: C5 ED 7A D8 EC 02 5F F2 EA 

Decoded:
Key      : C5
b1       : 25
ID       : 5A92
Command  : 02
Argument : CB
Sequence : 6C
Group    : 05
Checksum : 3C
fut089 packet received (9 bytes):
Raw packet: 9E 09 6D CC EF F8 2A B2 33 

Decoded:
Key      : 9E
b1       : 25
ID       : 5A92
Command  : 02
Argument : D8
Sequence : 69
Group    : 05
Checksum : 2B
fut089 packet received (9 bytes):
Raw packet: AF 40 AC 7E B6 F0 E8 CB 45 

Decoded:
Key      : AF
b1       : 25
ID       : 5A92
Command  : 02
Argument : DE
Sequence : 67
Group    : 05
Checksum : 20
fut089 packet received (9 bytes):
Raw packet: 0B E4 08 6A A2 16 0A AF 1D 

Decoded:
Key      : 0B
b1       : 25
ID       : 5A92
Command  : 02
Argument : E8
Sequence : 65
Group    : 05
Checksum : 14
fut089 packet received (9 bytes):
Raw packet: 04 D4 1B A5 57 5A D2 5E 19 

Decoded:
Key      : 04
b1       : 25
ID       : 5A92
Command  : 02
Argument : EA
Sequence : 64
Group    : 05
Checksum : 12
fut089 packet received (9 bytes):
Raw packet: 0C CC 23 9D 4F 66 C5 56 10 

Decoded:
Key      : 0C
b1       : 25
ID       : 5A92
Command  : 02
Argument : EE
Sequence : 63
Group    : 05
Checksum : 0D
fut089 packet received (9 bytes):
Raw packet: E0 F8 F7 71 43 7E F8 42 F3 

Decoded:
Key      : E0
b1       : 25
ID       : 5A92
Command  : 02
Argument : F2
Sequence : 62
Group    : 05
Checksum : 04
fut089 packet received (9 bytes):
Raw packet: 1A 8D E9 58 7B 4F 71 36 C1 

Decoded:
Key      : 1A
b1       : 25
ID       : 5A92
Command  : 02
Argument : 03
Sequence : 5C
Group    : 05
Checksum : D5
fut089 packet received (9 bytes):
Raw packet: 1E 89 ED 4C 6F 47 7C 32 D4 

Decoded:
Key      : 1E
b1       : 25
ID       : 5A92
Command  : 02
Argument : 07
Sequence : 5B
Group    : 05
Checksum : CC
fut089 packet received (9 bytes):
Raw packet: 7B 74 78 DA 32 04 A6 3F E7 

Decoded:
Key      : 7B
b1       : 25
ID       : 5A92
Command  : 02
Argument : 0E
Sequence : 59
Group    : 05
Checksum : BE
fut089 packet received (9 bytes):
Raw packet: 90 48 A7 21 D3 20 74 D2 F1 

Decoded:
Key      : 90
b1       : 25
ID       : 5A92
Command  : 02
Argument : 20
Sequence : 56
Group    : 05
Checksum : B6

GROUP 5 ON (driver connected)

fut089 packet received (9 bytes):
Raw packet: 95 1D 4A A8 DD C0 67 E2 A9 

Decoded:
Key      : 95
b1       : 25
ID       : 5A92
Command  : 01
Argument : 05
Sequence : 54
Group    : 05
Checksum : 4D

sidoh, 2020-01-02:
Thanks!

This all looks pretty normal.  Are you saying that you don't see packets show up for groups 1-4?  I think I might not be fully understanding the problem you're seeing.

(btw - doesn't matter if the bulbs are paired or not; the remotes have no knowledge whether anything is paired with them)

l1q1d, 2020-01-03:
Hi Sidoh,
I haven't added the log but the packet from 1-4 are normal.
The problema I see is that in the actual interface we are limited to 4, instead we should be able to see up to 8. (see the picture)
[image]

garmck, 2020-01-03:
Change the remote type to FUT089

sidoh, 2020-01-03:
Yep, @garmck's right (thanks garmck :).

Gonna close.

closed 2020-01-03 by sidoh