sidoh/esp8266_milight_hub #576: Incorrect State reports cause Home Assistant color jumps

Narquadah, 2019-12-31:
### Describe the bug
Using Home Assistant with state updates enabled causes "erratic" behavior see GIF.
[image]

Observing mqtt behaviour:  
HA color change:  
`homeassistant/light/milight_hub_540774/rgb_cct_0x1_0/config`
```json
{
  "schema" : "json",
  "name" : "Unten",
  "command_topic" : "milight/0x1/rgb_cct/0",
  "state_topic" : "milight/states/0x1/rgb_cct/0",
  "device" : {
    "manufacturer" : "esp8266_milight_hub",
    "sw_version" : "1.10.5",
    "identifiers" : [ 540774, 1, "rgb_cct", 0 ]
  },
  "availability_topic" : "milight/client_status",
  "payload_available" : "connected",
  "payload_not_available" : "disconnected",
  "brightness" : true,
  "effect" : true,
  "effect_list" : [ "night_mode", "white_mode", "0", "1", "2", "3", "4", "5", "6", "7", "8" ],
  "rgb" : true,
  "color_temp" : true
}
```

`milight/0x1/rgb_cct/0`
```json
{
  "state" : "ON",
  "color" : {
    "r" : 255,
    "g" : 0,
    "b" : 0
  }
}
```
Received:
`milight/update/0x1/rgb_cct/0`
```json
{
  "state" : "ON"
}
```

`milight/states/0x1/rgb_cct/0`
```json
{
  "state" : "ON",
  "brightness" : 255,
  "bulb_mode" : "color",
  "color" : {
    "r" : 200,
    "g" : 255,
    "b" : 145
  }
}
```

`milight/update/0x1/rgb_cct/0`
```json
{"hue":0}
```

`milight/states/0x1/rgb_cct/0`
```json
{
  "state" : "ON",
  "brightness" : 255,
  "bulb_mode" : "color",
  "color" : {
    "r" : 255,
    "g" : 145,
    "b" : 145
  }
}
```

`milight/update/0x1/rgb_cct/0`
```json
{"saturation":100}
```

`milight/states/0x1/rgb_cct/0`
```json
{
  "state" : "ON",
  "brightness" : 255,
  "bulb_mode" : "color",
  "color" : {
    "r" : 255,
    "g" : 0,
    "b" : 0
  }
}
```

In this case the result is the same as the input; however that is not always the case.
```json
{
  "state" : "ON",
  "color" : {
    "r" : 255,
    "g" : 0,
    "b" : 255
  }
}
```
results in 
```json
{
  "state" : "ON",
  "brightness" : 255,
  "bulb_mode" : "color",
  "color" : {
    "r" : 255,
    "g" : 0,
    "b" : 250
  }
}
```

------------------------------------------

Manual publish 
`milight/0x1/rgb_cct/0`
```json
{
  "state": "ON",
  "color": {
    "r": 255,
    "g": 0,
    "b": 0
  }
}
```
Received:
`milight/update/0x1/rgb_cct/0`
```json
{"state":"ON"}
```

`milight/states/0x1/rgb_cct/0`
```json
{
  "state": "ON",
  "brightness": 255,
  "bulb_mode": "color",
  "color": {
    "r": 0,
    "g": 250,
    "b": 255
  }
}
```

`milight/update/0x1/rgb_cct/0`
```json
{"hue":0}
```

`milight/states/0x1/rgb_cct/0`
```json
{
  "state": "ON",
  "brightness": 255,
  "bulb_mode": "color",
  "color": {
    "r": 255,
    "g": 0,
    "b": 0
  }
}
```

`milight/update/0x1/rgb_cct/0`
```json
{"saturation":100}
```

`milight/states/0x1/rgb_cct/0`
```json
{
  "state" : "ON",
  "brightness" : 255,
  "bulb_mode" : "color",
  "color" : {
    "r" : 255,
    "g" : 0,
    "b" : 0
  }
}
```

### Steps to reproduce

1. Enable state mqtt with `milight/states/:device_id/:device_type/:group_id`
2. Use HomeAssistant MQTT Discovery
3. Change color in HA

### Expected behavior

Milight Hub responds with the correct state (or without in between states)

### Setup information

#### Firmware version

#### Output of http://milight-hub.local/about

```json
{
  "firmware": "milight-hub",
  "version": "1.10.5",
  "ip_address": "x.x.x.x",
  "reset_reason": "External System",
  "variant": "d1_mini",
  "free_heap": 17576,
  "arduino_version": "2_4_2",
  "queue_stats": {
    "length": 0,
    "dropped_packets": 0
  }
}
```

#### Output of http://milight-hub.local/settings

```json
{
  "admin_username": "",
  "admin_password": "",
  "ce_pin": 4,
  "csn_pin": 15,
  "reset_pin": 0,
  "led_pin": -2,
  "radio_interface_type": "nRF24",
  "packet_repeats": 50,
  "http_repeat_factor": 1,
  "auto_restart_period": 0,
  "mqtt_server": "homeassistant.local:1883",
  "mqtt_username": "xxxxx",
  "mqtt_password": "xxxxxx",
  "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
  "mqtt_update_topic_pattern": "milight/update/:device_id/:device_type/:group_id",
  "mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
  "mqtt_client_status_topic": "milight/client_status",
  "simple_mqtt_client_status": true,
  "discovery_port": 48899,
  "listen_repeats": 3,
  "state_flush_interval": 10000,
  "mqtt_state_rate_limit": 0,
  "packet_repeat_throttle_sensitivity": 0,
  "packet_repeat_throttle_threshold": 500,
  "packet_repeat_minimum": 3,
  "enable_automatic_mode_switching": false,
  "led_mode_wifi_config": "Fast toggle",
  "led_mode_wifi_failed": "On",
  "led_mode_operating": "On",
  "led_mode_packet": "Flicker",
  "led_mode_packet_count": 3,
  "hostname": "milight-hub",
  "rf24_power_level": "MAX",
  "rf24_listen_channel": "LOW",
  "wifi_static_ip": "",
  "wifi_static_ip_gateway": "",
  "wifi_static_ip_netmask": "",
  "packet_repeats_per_loop": 10,
  "home_assistant_discovery_prefix": "homeassistant",
  "wifi_mode": "n",
  "default_transition_period": 500,
  "rf24_channels": [
    "LOW",
    "MID",
    "HIGH"
  ],
  "device_ids": [
    1
  ],
  "gateway_configs": [],
  "group_state_fields": [
    "state",
    "brightness",
    "mode",
    "color_temp",
    "bulb_mode",
    "computed_color"
  ],
  "group_id_aliases": {
    "Unten": [
      "rgb_cct",
      1,
      0
    ]
  }
}
```

### Additional context
For now disabling state updates mitigates the issue.

sidoh, 2019-12-31:
Hello, and thanks for the detailed report!

For more context, there are two things going on here:

1. **Lossy conversion from RGB**. Milight colors are specified with a hue on a [0,255] scale and a saturation on a [0,100] scale, so the conversion from RGB is lossy.  I'll spare you the details, but for a variety of reasons that make good technical sense, state is persisted and reported in terms of the values transmitted to the bulbs.  The loss comes from converting from RGB to HSL, and then back to RGB.
2. **Color updates are not atomic**.  Hue and saturation commands must be sent separately.  State updates are sent for each of them.  State updates can be throttled, but that doesn't really solve this problem.

(1) isn't really solvable without some significant technical sacrifices that I don't think I'm willing to make.  However...

The expected behavior you included makes sense:

> Milight Hub responds with the correct state (or without in between states)

and is mostly solved by allowing users to configure a [debounce](https://css-tricks.com/debouncing-throttling-explained-examples/) interval.

Long story short -- (2) is mostly solvable with a debounce interval.  (1) is unfortunately not addressable.  Thoughts?

Narquadah, 2020-01-02:
Hey, thanks for your comprehensive answer!

The debounce seems like a good solution to reduce the jumpiness. However, would it be possible to get HA to send an origin with the updated color? If yes sending updates from HA could be could possibly done without publishing a state or if the state would declare which application/origin it was triggered by HA could ignore states published by itself.

As for the lossy conversion would the loss be lower if HA used HSV instead of the RGB values? While there would be some loss, as Home Assistance HSV uses a 0 to 360 Range, it might improve accuracy.

sidoh, 2020-01-02:
espMH uses the builtin [light.mqtt](https://www.home-assistant.io/integrations/light.mqtt/#hs-color) component to integrate with HASS, so not really possible to have the sort of conditional logic you're describing without splitting off of that (which I don't want to do without a really strong reason).

If you don't want espMH to track state at all, you can turn on [optimistic mode](https://www.home-assistant.io/integrations/light.mqtt/#optimistic) within HASS or just leave off the state topic.  State tracking is important if you're controlling lights outside of HASS (like the espMH web UI, a physical remote, etc.), but not really otherwise.  

Using [HS colors](https://www.home-assistant.io/integrations/light.mqtt/#hs-color) would probably help, but the main problem is the rounding.  I'll open a separate issue to add support for HS fields.

sidoh, 2020-04-14:
Closing in favor of #615 (same underlying issue)

closed 2020-04-14 by sidoh