sidoh/esp8266_milight_hub #579: Pairing on second Device ID

quadragin, 2020-01-03:
I cloned my existing Limitless LED hub and was able to pair a new LED strip controller.
I created a second Device ID under the UDP in settings called 0x1.
I added the new 0x1 and a new alias for group 1 made sure RBGW was selected.
followed the paring procedure and the controller will not pair.

sidoh, 2020-01-03:
Think I may not be following exactly what you're doing, but if you're trying to pair with a device that's already been paired, you may need to unpair it first.

quadragin, 2020-01-03:
I found out what I did wrong. my device 0xE04 and 0x1 I set up with the same port. I changed UDP to
0x046 port 59877 and 0x1 port 8900 and was able to pair. After pairing however the hub has seemed to stop working. No control of lights on ether device...

quadragin, 2020-01-03:
actually I was able to change them to flash mode but now they are stuck in flash mode, I can no longer turn them off or change color. With the Web interface or Home Assistant.

quadragin, 2020-01-03:
Well I rebooted the hub, deleted the 0x1. was able to gain control of lights on 0x046 by hitting white button twice. recreated 0x1 and all seems to be working now.

closed 2020-01-03 by quadragin