sidoh/esp8266_milight_hub #581: how do I add more than 4 groups in HASSIO

quadragin, 2020-01-03:
I have now set up my 5th group... Second device id / group 1
I do not see anywhere how to set up my second device in HA
Here is my current lights.yaml file. Working fine.

- platform: limitlessled
  bridges:
    - host: 192.168.107.20
      groups:
      - number: 1
        type: rgbw
        fade: true
        name: Railing
      - number: 2
        type: rgbw
        fade: true
        name: Stairs
      - number: 3
        type: rgbw
        fade: true
        name: Floor
        fade: true
      - number: 4
        type: rgbw
        fade: true
        name: Pool

garmck, 2020-01-04:
Each Device has a specific UDP port, you will need to define the ports in your configuration.

TBH, you'd be est placed to ditch UPD/Limitless integration and start using MQTT, you'll get a vastly better experience.

quadragin, 2020-01-04:
Agreed, Ive just started reading on how to do that.. I think I have my hub mqtt info correct.
Should the lights show up automatically or do I need to create an entry for each one in my lights.yaml or configuration.yaml?
[image]

garmck, 2020-01-04:
discovery will work if you have specified the config in HASS, i have the following
`discovery: true`
`discovery_prefix: homeassistant`

But I don't use discover for the lights, I use a specific yaml for each light group/room
lounge, kitchen, exterior etc

quadragin, 2020-01-04:
can you show me a sample of one of your lights, so I can reverse engineer.
I'm a NOOB to HA

thanks in advance

sidoh, 2020-01-04:
There are some examples on the wiki:

https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant

poudenes, 2020-01-05:
I added all my lights as a individual bulb in HA.:

---
```
platform: mqtt
name: "Toilet 1"
schema: json
command_topic: "milight/0xAAAE/rgb_cct/1"
state_topic: "milight/states/0xAAAE/rgb_cct/1"
qos: 1
optimistic: "false"
retain: "true"
brightness: "true"
rgb: "true"
color_temp: "true"
effect: "true"
effect_list: !include /config/package/templates/light_effects.yaml
```
```
"milight/0xAAAE/rgb_cct/1" group 1
"milight/0xAAAE/rgb_cct/2" group 2
"milight/0xAAAE/rgb_cct/3" group 3
"milight/0xAAAE/rgb_cct/4" group 4
"milight/0xAAAE/rgb_cct/0" All groups
```

I created many virtual remotes. Every virtual remote for me is a room
Every virtual remote (room) have 4 groups and a all group
I add in every group 1 bulb. And can control them all together with group all
in HA you can control every bulb individually or as group.

totally free and no limits!!

quadragin, 2020-01-05:
Thanks
Ill get to groups and effects once I can add a light card and turn a light on and off in HA...
did you have to specify your MiLIght Hub somewhere?

quadragin, 2020-01-05:
STILL no luck, the only light entities that show up to configure are the LimitlessLED
Here is my Lights.yaml
- platform: limitlessled
  bridges:
    - host: 192.168.107.20
      groups:
      - number: 1
        type: rgbw
        fade: true
        name: Railing
      - number: 2
        type: rgbw
        fade: true
        name: Stairs
      - number: 3
        type: rgbw
        fade: true
        name: Floor
        fade: true
      - number: 4
        type: rgbw
        fade: true
        name: Pool
        
- platform: mqtt
  name: "Floor"
  schema: json
  command_topic: "milight/0xE046/rgbw/3"
  state_topic: "milight/0xE046/rgbw/3"
  qos: 1
  optimistic: "false"
  retain: "true"
  brightness: "true"
  rgb: "true"
  color_temp: "true"
  effect: "true"

poudenes, 2020-01-05:
Read info about platform mqtt
I have split.up my configuration.
Hub is not in HA at all. Because the lights will be controlled using mqtt

poudenes, 2020-01-05:
https://www.home-assistant.io/integrations/light.mqtt/

See example 
```
light:
  - platform: mqtt
    name: "Office Light RGB"
    state_topic: "office/rgb1/light/status"
    command_topic: "office/rgb1/light/switch"
    brightness_state_topic: "office/rgb1/brightness/status"
    brightness_command_topic: "office/rgb1/brightness/set"
    rgb_state_topic: "office/rgb1/rgb/status"
    rgb_command_topic: "office/rgb1/rgb/set"
    state_value_template: "{{ value_json.state }}"
    brightness_value_template: "{{ value_json.brightness }}"
    rgb_value_template: "{{ value_json.rgb | join(',') }}"
    qos: 0
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false light:
  - platform: mqtt
    name: "Office Light RGB"
    state_topic: "office/rgb1/light/status"
    command_topic: "office/rgb1/light/switch"
    brightness_state_topic: "office/rgb1/brightness/status"
    brightness_command_topic: "office/rgb1/brightness/set"
    rgb_state_topic: "office/rgb1/rgb/status"
    rgb_command_topic: "office/rgb1/rgb/set"
    state_value_template: "{{ value_json.state }}"
    brightness_value_template: "{{ value_json.brightness }}"
    rgb_value_template: "{{ value_json.rgb | join(',') }}"
    qos: 0
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
```

quadragin, 2020-01-06:
OK, so HA sends a command in mqtt, the mqtt broker then handles the rest.
In these example above the only thing that is still needed is the unique_id: "milight/0xE046/rgbw/3"
of each bulb or strip controller.

poudenes, 2020-01-06:
Try this:

```
light:
  - platform: mqtt
     name: "Toilet 1"
     schema: json
     command_topic: "milight/0xE046/rgbw/3"
     state_topic: "milight/0xE046/rgbw/3"
     qos: 1
     optimistic: "false"
     retain: "true"
     brightness: "true"
     rgb: "true"
     color_temp: "true"
     effect: "true"
```

quadragin, 2020-01-06:
I swapped out my previous with this one.
The entity now shows as light.theater_2, again with no response.
When I touch the slider to turn the light on the slider moves to on for a second then switches back to off...

sidoh, 2020-01-06:
Command and state topics should always be different:

```yaml
     command_topic: "milight/0xE046/rgbw/3"
     state_topic: "milight/0xE046/rgbw/3"
```

should be something like:

```yaml
     command_topic: "milight/command/0xE046/rgbw/3"
     state_topic: "milight/state/0xE046/rgbw/3"
```

Make sure they match your MQTT settings in espMH.

sidoh, 2020-01-06:
Please also try following the documentation on the wiki.  It should be comprehensive:

https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant

poudenes, 2020-01-06:
Yes my fault... both must be different. Didn't post correct code... Hope it will work now for you!

quadragin, 2020-01-07:
I have read through that many times, 
I guess to me confusing part is where he states  :device_id then do I have mine like this :0x0E46 or 0x0E46?
What requires the semicolon and quotes and when are they not required.
I can now see my one light I have configured as light.floor1 but HA says its unavailable.
I wish there was an easy way to test the mosquitto broker but it looks to me it is working OK.

sidoh, 2020-01-07:
You should be able to pretty much copy/paste stuff from the wiki, @quadragin.  No need to specialize.

There's more detailed documentation on the MQTT pattern structure in [the MQTT section of the README](https://github.com/sidoh/esp8266_milight_hub#more-detail-on-mqtt_topic_pattern) if you're wanting to get a deeper understanding.

When you tell espMH to listen for commands on the **topic pattern** `milight/commands/:device_id/:device_type/:group_id`, it will _actually_ listen on the wildcard topic `milight/commands/+/+/+` (`+` is a [single-level wildcard](https://mosquitto.org/man/mqtt-7.html)).  Then, when it receives a message on the topic `milight/commands/0x1234/rgb_cct/1`, it will know that the **Device ID** is `0x1234`, the **Device Type** is `rgb_cct`, and the **Group ID** is `1`, because the stand-in tokens appear in that order.

The easiest way to check if messages are coming through your broker is to attach a debug client to it.  I use `mosquitto_sub` like so:

```
$ mosquitto_sub -h HOSTNAME -u USERNAME -P PASSWORD -t 'milight/#' -v
```

quadragin, 2020-01-08:
Here is what is returned back. And as I turn lights on and off or change mode in 
esp8266_milight_hub web client I see the commands on the debug.
So I think all is working through the Mosquitto broker
login as: root
root@192.168.107.30's password:

  _    _                 _
 | |  | |               (_)
 | |__| | __ _ ___ ___   _  ___
 |  __  |/ _` / __/ __| | |/ _ \
 | |  | | (_| \__ \__ \_| | (_) |
 |_|  |_|\__,_|___/___(_)_|\___/

Our Cli:
$ hassio help

~ $ mosquitto_sub -h 192.168.107.30 -u .....-P .....-t 'milight/#' -v

milight/client_status connected
milight/states/0xE046/rgbw/3 {"state":"OFF","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/1 {"state":"OFF","brightness":245,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/4 {"state":"ON","brightness":0,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/2 {"state":"ON","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/0 {"state":"ON","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/3 {"state":"ON","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/3 {"state":"OFF","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/0 {"state":"ON","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/0 {"state":"OFF","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/3 {"state":"OFF","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/client_status connected
milight/states/0xE046/rgbw/3 {"state":"OFF","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/1 {"state":"OFF","brightness":245,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/4 {"state":"ON","brightness":0,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/2 {"state":"ON","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/0 {"state":"ON","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/3 {"state":"ON","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/3 {"state":"OFF","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}
milight/states/0xE046/rgbw/0 {"state":"ON","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/0 {"state":"OFF","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
milight/states/0xE046/rgbw/3 {"state":"OFF","brightness":0,"bulb_mode":"color","color":{"r":106,"g":0,"b":255}}

sidoh, 2020-01-10:
If I'm reading what you're saying, everything is working as expected when you're controlling through the UI, meaning espMH is configured correctly.  

If something else is not working (e.g., HASS), that piece is not configured correctly.

quadragin, 2020-01-10:
SO I have tried just about everything except uninstalling Mosquitto and Zigbee2MQTT and starting over.
If in putty I run the listing Mosquitto command; mosquitto_sub -h 192.168.107.30 -u uname -P passw -t "#" -v
I get responses, my z2m coordinator is talking my z2m router is talking my MiLight hub and bulbs are talking.
IF i go to Home Assistant UI Developer/MQTT Tab/Listen to Topic /listen to ether milight/# or just # nothing happens below the Listen to a topic card.

I can however get my MiLight to work with Home Assistant using UDP.

any last ideas before I start over? I may even re-flash Hass.io on my RasPI and completely start from scratch.

garmck, 2020-01-12:
I have my lights set up as follows:
```
- platform: mqtt
  schema: json
  name: "Lounge"
  state_topic: "milight/state/0x380A/rgb_cct/1"
  command_topic: "milight/command/0x380A/rgb_cct/1"
  brightness: true
  color_temp: true
  rgb: true
  effect: true
  effect_list:
    - white_mode
    - night_mode
    - 0
    - 1
    - 2
    - 3
    - 4
    - 5
    - 6
    - 7
    - 8
```
Make sure you have the state and command set in the milight hub

sidoh, 2020-01-12:
Not sure where else to point ya, @quadragin.  The configurations outlined in the documentation work for me and several others.  From what you've said, it sounds like HASS isn't publishing to MQTT correctly, so that's probably where I'd focus my efforts.

quadragin, 2020-01-20:
**@garmck @sidoh I have started from scratch rebuilt my HASSOS.
I have Mosquitto Broker installed and running with a user name/password and access control list enabling unrestricted access to all topics.

The Milights are not auto added in the HA/Overview screen, nor does it show up in Configuration/Devices.

In a terminal session I run mosquitto_sub -h 192.168.107.30 -u .....-P .....-t 'milight/#' -v
I get this back in return when I issue a command in the espMH web interface
milight/states/0xE046/rgbw/2 ["state: "ON", "brightness": 20, "bulb_mode...etc..

In the espMH**
MQTT topic pattern - milight/:device_id/:device_type/:group_id
MQTT update topic pattern - nothing
MQTT state topic pattern - milight/states/:device_id/:device_type/:group_id

**and have added this to my configuration.yaml**
mqtt:
  broker: 192.168.107.30
  discovery: true
  discovery_prefix: homeassistant

light:
  - platform: mqtt
    schema: json
    name: "Deck_Stairs"
    state_topic: "milight/state/0xE046/rgbw/2"
    command_topic: "milight/command/0xE046/rgbw/2"
    brightness: true
    color_temp: true
    rgb: true
    effect: true
    effect_list:
      - white_mode
      - night_mode
      - 0
      - 1
      - 2
      - 3
      - 4
      - 5
      - 6
      - 7
      - 8

**My Mosquitto log looks like this:**
[10:57:27] INFO: Setup mosquitto configuration
[10:57:27] INFO: Found local users inside config
[10:57:27] INFO: Initialize Hass.io Add-on services
[10:57:27] INFO: Initialize Home Assistant discovery
[10:57:28] INFO: Start Mosquitto daemon
1579543048: mosquitto version 1.6.3 starting
1579543048: Config loaded from /etc/mosquitto.conf.
1579543048: Loading plugin: /usr/share/mosquitto/auth-plug.so
1579543048:  ├── Username/password checking enabled.
1579543048:  ├── TLS-PSK checking enabled.
1579543048:  └── Extended authentication not enabled.
1579543048: |-- *** auth-plug: startup
1579543048: Opening ipv4 listen socket on port 1883.
1579543048: Opening ipv6 listen socket on port 1883.
1579543048: Opening websockets listen socket on port 1884.
1579543048: Opening ipv4 listen socket on port 8883.
1579543048: Opening ipv6 listen socket on port 8883.
1579543048: Opening websockets listen socket on port 8884.
1579543048: Warning: Mosquitto should not be run as root/administrator.
1579543049: New connection from 192.168.107.20 on port 1883.
[INFO] found mqtt on local database
1579543050: New client connected from 192.168.107.20 as milight-hub-1670881 (p2, c1, k15, u'mqtt').
1579543088: New connection from 192.168.107.30 on port 1883.
1579543088: Socket error on client <unknown>, disconnecting.
1579543704: New connection from 172.30.32.1 on port 1883.
[INFO] found mqtt on local database
1579543705: New client connected from 172.30.32.1 as mosq-P7wey5hTiNNWZ390EI (p2, c1, k60, u'mqtt').

quadragin, 2020-01-20:
Im going to close this and open a new topic as this issue has migrated away from more that 4 UDP groups

closed 2020-01-20 by quadragin