sidoh/esp8266_milight_hub #582: version 1.10.5 (nodemcuv2) unresponsive

quadragin, 2020-01-04:
### Describe the bug
using the web interface when i select a device and try to change the color the bulb does not respond most of the time.

### Steps to reproduce

### Expected behavior

### Setup information
2 device ids
 RBGW 0xE046 UDP Port 5987 with 4 groups configured
 RBGW 0x1 UDP Port 8900 with 1 group configured

#### Firmware version

#### Output of http://milight-hub.local/about
{"firmware":"milight-hub","version":"1.10.5","ip_address":"192.168.107.20","reset_reason":"External System","variant":"nodemcuv2","free_heap":17192,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}

#### Output of http://milight-hub.local/settings
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":60,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"g","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[1,57414],"gateway_configs":[[57414,5987,6],[1,8900,6]],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"Deck_Rail":["rgbw",57414,1],"Deck_Stairs":["rgbw",57414,2],"Floor":["rgbw",57414,3],"Pool":["rgbw",57414,4],"Theater":["rgbw",1,1],"t":["rgbw",1,1]}}

### Additional context

quadragin, 2020-01-04:
more testing shows that the issue is mainly with the bulbs and not the LED strip controllers.
once the bulbs are in flash mode they do not like go back to normal colors or white.

sidoh, 2020-01-04:
I can't reproduce this, so will need more information to be of any help.  

Things to try:

* Look at serial logs (`pio device monitor` while ESP8266 is plugged into your computer)
* Enable debug logging (compile with `-D DEBUG_PRINTF`) and check serial logs again
* Check sniffed packets while bulbs are not responding

jellewie, 2020-01-07:
Could this be interference? When I steam 100Mbps though the air locally to my Steam link at 2.4Ghz, the lights show similar problems receiving a signal.

On a small side note; I've tried using garbage power supplies when the antenna is in full power mode the power rail dropped a bit, and some packages were getting lost. Can you try to reproduce this with a 0.5A+ power adapter for your ESP? (just for good measure, I don't expect it would fix it)

sidoh, 2020-01-07:
Definitely possible.  I get interference problems if the nRF and ESP8266 are too close together.

quadragin, 2020-01-08:
They are not close together, but I have the ESP plugged into my PC USB and seems to be really reliable.
I will pick up a quality .5+amp supply and try put the ESP where I had it before

quadragin, 2020-01-12:
Still having issues, not sure if its power or interference. just purchased some quality 2amp power supplies. 
Unit powers up flashes normal, flickers when commands are sent, but no response from devices.
Plug back into the USB hub plugged into my PC and at times it work perfectly and other times some devices resond and others do not. 
I have played around with the radio settings. and maybe now have hosed them up.
What are the default radio settings?
and how can I go back to 1.10.4 I think that was the previous flash.

thanks

jellewie, 2020-01-13:
Not 100% the default settings, but close https://i.imgur.com/okePyJ4.png (their mine and they work). 
The real default values can be this file I believe [/blob/master/lib/Settings/Settings.cpp](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Settings/Settings.cpp).

to go back to 1.10.4 you can just download the old compiled bin file [here](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.10.4). and [upload](https://github.com/sidoh/esp8266_milight_hub/issues/584#issuecomment-571538374) it ("Admin - Upgrade firmware" can be used if you can access that)

quadragin, 2020-01-20:
Thanks
I purchased a NON USB power supply. its a RasPi type 2.5 amp with a filter and that seems to have done the trick.
thank you for the posts of the older firmares

quadragin, 2020-01-20:
I purchased a non usb power supply 2.5 amp with a filter that seems to have done the trick

closed 2020-01-20 by quadragin