sidoh/esp8266_milight_hub #584: version 1.10.5 (nodemcuv2) broken

jellewie, 2020-01-05:
### Describe the bug

I've broken my ESP8266 with the update (steps below)
but for good practice, I got a new one and fashed the binary to it. It behaved weird, and spit out errors.

**some images**
https://i.imgur.com/cRkENQC.png
https://i.imgur.com/GHAV78U.png

**Serial log**
https://pastebin.com/f2yCpneC

### Steps to reproduce

I had .3 updated with portal to .4. and now a long time later I tried to update to .5 with the web portal. The ESP did never get online. I flashed 1.10.5 again but with cable, but it's unable to save or load settings. 

### Expected behavior

That an update worked, and it doesn't crash

### Setup information

NodeMCU ESP8266 (2x)

#### Firmware version

Updated 1.10.4 to 1.10.5 on a NodeMCU ESP8266 (original it was a 1.10.3)
and flashed 1.10.5 to a clean NodeMCU ESP8266

#### Output of http://milight-hub.local/about

Can't do this, ESP is stuck in boot loop

#### Output of http://milight-hub.local/settings

Can't do this, ESP is stuck in boot loop

### Additional context

Might be related to #582

jellewie, 2020-01-05:
The old one (who has updated from .4 to .5) now has this serial log:

@9600
```
2DOz4⸮h⸮$⸮$⸮x⸮⸮:⸮lb<⸮l⸮D:hŀF⸮⸮

NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4
lua: cannot open init.lua
> 
```

or 
@74880
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 28740, room 16 
tail 4
chksum 0xcd
load 0x3ffe8000, len 2888, room 4 
tail 4
chksum 0xbc
load 0x3ffe8b50, len 15252, room 4 
tail 0
chksum 0x4f
csum 0x4f
R??'R⸮H@!⸮
```

And then nothing else, ESP is stuck in some sort of while loop

9600, 2020-01-05:
You called?

sidoh, 2020-01-05:
Thanks for providing all of this info upfront!  Very helpful.

Your 2nd ESP8266 looks like it's running the NodeMCU Lua firmware, which is what NodeMCUs run from the factory.  Looks like whatever you did to flash espMH didn't take.

[Here's the diff from 1.10.4..1.10.5](https://github.com/sidoh/esp8266_milight_hub/compare/1.10.4..1.10.5).  I see nothing there that could cause this behavior, so I don't think the issue is in the .4 -> .5 version.  If there is a bug, my guess is there's an issue in the 1.10 branch that is getting triggered inconsistently by an update.

The serial logs you provided are interesting, especially this part:

```
ERROR[2]: Flash Erase Failed
ERROR[2]: Flash Erase Failed
ERROR[2]: Flash Erase Failed
ERROR[2]: Flash Erase Failed
ERROR[2]: Flash Erase Failed
```

I think this only happens when:

1. There's a hardware failure with the onboard flash
2. You're trying to write a firmware image that does not match the size of the flash chip you have

Is it possible you tried to flash the d1_mini_pro or esp07 image?

sidoh, 2020-01-05:
re: related issues you linked:

* #582 looks related in that the title is nearly the same, but is a completely separate issue.  the firmware is working, but bulbs are not responding to packets when the user expects them to.  i suspect this is not an actual bug)
* #530 is almost certainly related to a memory leak I've not been able to track down and does not cause the boot loop behavior you're seeing.

jellewie, 2020-01-05:
Hmm I can't really seem something indeed in the .4 to .5 diff indeed. I think your "triggered inconsistently by an update" is quite plausible

Here is a better description of my first steps:
1. checked for an update on the web ui. and downloaded the version it gave me (NodeMCU which I use)
2. uploaded it with the portal, but after 10min the hub still wasn't reachable through it's IP.
3. I rebooted (power off and on) the ESP which similar results,
4. I decided that it had broken, connected it with USB and downloaded 1.10.5 bin file from the GitHub page. I've used "[nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher)" to upload the sketch (forgot to change the flash size to 4mb the first time) but this eventually worked, 
5 the webpage could sometimes be reached, but all settings where blank, and after setting them they would be cleared a couple of seconds later (I was able to turn my light on for a few seconds)
6. After I tried to apply all settings, the ESP got stuck in the boot (I guess setting the pins??) both my ESP's have this issue, A simple blink and communication Arduino sketch still works. 

The "ERROR[2]: Flash Erase Failed" was with the 4mb flash size (I guess that was the right setting for the NodeMCU?) after I tried to update from 1.10.3 or 1.10.4 to a newer (precompiled) version. those all had the 'can't load settings error'

> Is it possible you tried to flash the d1_mini_pro or esp07 image?
no, I only used the NodeMCU version, it is however possible that a software (Arduino, nodemcu-flasher, PlatformIO, or ESP download tool) has change something in the flash that hasn't been cleared

jellewie, 2020-01-05:
I now have them both cleared with ESP8266 with a 1Mb bin file (with only 0x'FF') and after that I uploaded the precompiled "1.10.4.bin" file. 

after I reset the ESP, it got into AP mode, connected and set the WIFI stuff, and then I was in the settings, but when I change anything and press submit, the ESP crashes and gets stuck in a boot loop. (I need to change some pins)

Both have the same issue 

`<more logs>
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Wifi connected succesfully

Setup complete (version 1.10.4)
Unhandled websocket event: 10
<and here it crashed>
H!⸮ԅ⸮⸮DHB⸮*WM`

sidoh, 2020-01-06:
If it's crashing when you're trying to save settings, your SPIFFS settings are wrong.  The tool you're using to flash firmware is probably setting the partition sizes differently.

jellewie, 2020-01-07:
So I've been hard at work with @jespertheend, here at the results;

**Update corruption**
The .4 to .5 update somehow got corrupted, **Sadly I have no clue how**, but I hope you would be able to fix it in the future.

**Serial log feedback**
After flash start a Serial monitor (like the Arduino one) @ 9600. this will give you Serial log feedback, reboot the ESP with the button to see startup logs. if this is only gibberish then there is something wrong, change the baud rate higher and retry until to you can read it, see [this ](https://www.espressif.com/sites/default/files/documentation/esp8266_reset_causes_and_common_fatal_exception_causes_en.pdf)file for reset cause information. 
I had an error 4 (Hardware WDT reset) this is an internal hardware error; this was caused by that one of the pins to the antenna was wrong. Please check the connections (intern != extern numbering, [check pinout](https://github.com/sidoh/esp8266_milight_hub#Installing))
[This is an example of a good serial log](https://pastebin.com/Zd9wLqPS)

## How it did work at the end
This made all 3 ESP's work again, with 3 different ways. all functions are as far as I know working.

**(hard, slow, can't recommend)"nodemcu-flasher-master"**
This one had the wrong settings by default, 
1. Change Baudrate high (460800) (this is the upload speed and later also debug speed)
and change Flash size to 4MByte
2. Clear the flash by adding the 4x 1mb [blank files](http://www.pratikpanda.com/wp-content/uploads/2016/05/blank_1MB.zip) and their location in the config tab (same as in the 'flash_download_tools' below), and then start the Flash.
3. Now select only the bin and Flash that one

**(1 instal + quite easy) "flash_download_tools_v3.6.8.exe"**
1. Reset flash with [this ](http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/)file, and flash bin with [these ](https://i.imgur.com/LkaU8Xa.png)settings (do 4x the empty )
2. Now select only the bin and Flash that one (unselect 4x empty, select bin)

**(best, but need more installs) With CMD**
took hours to get programs working together, like python and PlatformIO. if you have similar problems; de-installed all, first install PlatformIO and then install the rest.

1.  
```
cd <your direction>\esptool-master
python esptool.py erase_flash

```
2.  
```
cd <your direction>\esp8266_milight_hub-master
platformio run -e nodemcuv2 --target upload
```

closed 2020-01-07 by jellewie

sidoh, 2020-01-07:
Glad you got it working, and thanks for following up.  

I'm not sure how OTA updates go bad, but it has happened to me before as well.  Can be on the lookout for upstream fixes, but I'm using pretty standard SDK features to support OTA updates.

jellewie, 2020-01-08:
I should have checked Serial, and written down the behavior when it was bad. 
When it ever happens again I will for sure try to get some more info for you!