sidoh/esp8266_milight_hub #585: No Connection to MQTT / MilightHub

TimoWerr, 2020-01-07:
### Describe the bug

I have a setup like it´s described on github with a nodemcu. The Hub works fine using a browser. I want a connection between FHEM and sp8266_milight_hub with MQTT - FHEM Always says "Connection Refused"

`2020.01.03 16:49:35 1: MiLightHubClient: Can't connect to 192.168.178.85:1884: Operation now in progress`

`2020.01.03 16:49:35 1: MiLightHubClient: Can't connect to 192.168.178.85:1884: 192.168.178.85: Connection refused (111)`

### Steps to reproduce

I saved the standard settings in MQTT on the Hub

### Expected behavior

A connection between FHEM and the HUB

### Setup information
As it´s explained on github with a NODEMCU

#### Firmware version
esp8266_milight_hub_nodemcuv2-1.10.5.bin
esp8266_milight_hub_nodemcuv2-1.10.4.bin
 esp8266_milight_hub_nodemcuv2-1.10.0.bin (Actual version)

#### Output of http://milight-hub.local/about

`firmware	"milight-hub"
version	"1.10.0"
ip_address	"192.168.178.85"
reset_reason	"Power on"
variant	"nodemcuv2"
free_heap	17624
arduino_version	"2_4_2"
queue_stats	
length	0
dropped_packets	0`

#### Output of http://milight-hub.local/settings

```
admin_username	""
admin_password	""
ce_pin	4
csn_pin	15
reset_pin	0
led_pin	-2
radio_interface_type	"nRF24"
packet_repeats	50
http_repeat_factor	1
auto_restart_period	0
mqtt_server	"192.168.178.85:1884"
mqtt_username	"admin"
mqtt_password	"admin"
mqtt_topic_pattern	"milight/:device_id/:device_type/:group_id"
mqtt_update_topic_pattern	"milight/updates/:hex_device_id/:device_type/:group_id"
mqtt_state_topic_pattern	"milight/states/:hex_device_id/:device_type/:group_id"
mqtt_client_status_topic	"milight/LWT"
simple_mqtt_client_status	false
discovery_port	48899
listen_repeats	3
state_flush_interval	10000
mqtt_state_rate_limit	500
packet_repeat_throttle_sensitivity	0
packet_repeat_throttle_threshold	200
packet_repeat_minimum	3
enable_automatic_mode_switching	false
led_mode_wifi_config	"Fast toggle"
led_mode_wifi_failed	"On"
led_mode_operating	"Slow blip"
led_mode_packet	"Flicker"
led_mode_packet_count	3
hostname	"milight-hub"
rf24_power_level	"MAX"
rf24_listen_channel	"LOW"
wifi_static_ip	""
wifi_static_ip_gateway	""
wifi_static_ip_netmask	""
packet_repeats_per_loop	10
home_assistant_discovery_prefix	""
wifi_mode	"n"
rf24_channels	
0	"LOW"
1	"MID"
2	"HIGH"
device_ids	
0	23828
gateway_configs	[]
group_state_fields	
0	"state"
1	"brightness"
2	"mode"
3	"color_temp"
4	"bulb_mode"
5	"computed_color"
group_id_aliases	
1OG	
0	"rgb_cct"
1	0
2	1
```

Any Idea what´s wrong in my case? It would be Great if someone could help me...

sidoh, 2020-01-07:
what is `192.168.178.85`?  judging from your espMH settings, this should be your MQTT broker.

TimoWerr, 2020-01-07:
> what is `192.168.178.85`? judging from your espMH settings, this should be your MQTT broker.

This is the IP of the Hub itself so i used it for MQTT Settings to...

TimoWerr, 2020-01-07:
> > what is `192.168.178.85`? judging from your espMH settings, this should be your MQTT broker.
> 
> This is the IP of the Hub itself so i used it for MQTT Settings to...

In Addition with a port

sidoh, 2020-01-07:
You need an MQTT broker.  This is a 3rd service that sits between espMH and FHEM and serves as a message bus.

Read more about MQTT here: https://learn.sparkfun.com/tutorials/introduction-to-mqtt/all

closed 2020-01-07 by sidoh