sidoh/esp8266_milight_hub #587: Not all lights repsond to command

Prinsie, 2020-01-12:
I checked the troubleshooting page but still I have issues. I can send commands to the lightbulbs. But the problem I have is that often not all lightbulbs respond. I have in one group 8 lightbulbs and always one light (ramdom) does not respond. In other groups with less lightsbulbs I am having the same issues.
Because I can control the lights I assume it is not a wiring problem or hardware fault. My guess is that in the settings page some parameters need the be finetuned? But I have no clue wat to change.

The 2.4ghz module I have is NRF24L01+

### What is the model number of the device you're trying to control?

The lightbulbs I have are
FUT011
FUT017
FUT019

### What firmware version(s) have you tried? 
esp8266_milight_hub_nodemcuv2-1.10.5.bin

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
esp82667 nodemcu v3

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
CCT

### Have you tried controlling the device with a physical remote?
Controlling with the remote works just fine for all groups and lightbulbs

 

The remote I have is the FUT006.

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
{"firmware":"milight-hub","version":"1.10.5","ip_address":"192.168.1.108","reset_reason":"External System","variant":"nodemcuv2","free_heap":18192,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### /settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":20,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[8370],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{"1":["cct",8370,1],"2":["cct",8370,2],"3":["cct",8370,3],"4":["cct",8370,4],"all":["cct",8370,0]}}"
```

sidoh, 2020-01-12:
The only parameter that's going to help here is _Packet repeats_:

[image]

Other tips:

1. Try a different nRF.  Performance varies from batch to batch in my experience.
1. Make sure the ESP8266 and nRF are separated from each other by a bit.  They can interfere with each other.
1. Make sure you're using a good power supply.

jellewie, 2020-01-13:
@Prinsie can you do the remote test while the remote is stuck against where the ESP and nRF is?

**Side note on tip 2** of @sidoh 
If you want your ESP to be closed to your nRF (for like a compact design case) you must shield the ESP. the easiest way to block the unwanted Rf (from high-frequency things like the PSU, WIFI and BLE module) is with something metal, for example like @h4nc dit with it's [PCB](https://github.com/sidoh/espmh_pcb) Although probably by accident, but some distance or aluminum foil (make sure not to short things! and leave room for the WIFI module to breath)

**As a test;**
I taped my ESP32, the rNF to my NighHawk R7000 router
The light show very similar problems (they ignore ~75% of commands at 2M line of sight) package repeats seem to brute force 'solve' this problem. But the hub is then getting slowed down a lot.

I moved the ESP32 about 0.5M away from the router, the rNF is 1 jumper wire (10CM) away from the ESP, no other electric components at a radius of .5M from the ESP.
This resulting in all light behaving normally. I could even control lights (~95%+ good) at 150M away with only 1 concrete wall (20CM) further only line of sight.

**Side note on tip 3;**
I can measure a 0.12A at 4.89V which is about 0.6W. this is an average, not a peak! the nRF probably has a high peak power. The [datasheet](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf) of the nRF states a max of 13.5mA at 3V, but that is only the nRF and not sure how reliable that is.
But I do have bad 0.5A power supplies where this ESP isn't able to work with, the voltage line drops too much I guess, so the amplitude drops and the signal is messed up?

Prinsie, 2020-01-14:
Playing with the packet repeats improved things a bit but didn't solve it unfortunatly

I tried different power supplies didn't make a difference. 

The distance between the two modules is about 10 cm, should be fine I think.

I triend with the remote stuck against the nRF and ESP but the remote works fine. I think I just have a bad nRF module. I ordered some new onces on aliexpress. This time a more powerfull one, hopefully this works :-) : https://nl.aliexpress.com/item/1660900248.html?spm=a2g0s.9042311.0.0.45f14c4dkdwbWM

jellewie, 2020-01-15:
A good one to share the link, although it might also be a batch problem, but who knows.

Me and a friend used [this aliexpress](https://www.aliexpress.com/item/33055955341.html?spm=a2g0s.9042311.0.0.27424c4dhRPdA1) module, exactly the same specs and thing, just a different store (They worked fine for us). And all my test and conclusions are made with that one.

I can't see any reason why the link you provided should be different in any way, except that my link states an extra 100M range, but the rest of the text is just copied and PCB's look the same