sidoh/esp8266_milight_hub #589: Bulbs No Longer Responding

MatthewT1, 2020-01-18:
I've gone through the troubleshooting guide but am still having problems - 

We set this up about a year ago, and it had been working great until I had to turn off he hub and turn it back on. Our hub controls 5 lights - 2 CW/WM bulbs in our office, 1 CWRGB in the living room, and 2 CWRGB in the bedroom. We have 1 remote that's paired with the office ones. 

We had these communicating with Home Assistant over an MQTT server. 

- Confirmed that the Hub is picking up commands from the remote
- Confirmed that MQTT is working correctly
- Tested out controlling the bulbs from the Hub website directly, but they don't respond to them. 

I'm thinking there' s an issue between the hub and the bulbs, but I'm not sure how to reset them and try to re-pair them. 

This is the config we were using for Home Assistant: 

```yaml
light: 
  - name: Office
    # The command topic is derived by substituting real values for the
    # tokens in the setting mqtt_topic_pattern (e.g., "rgb_cct" for
    # ":device_type"):
    #                        ______________ Device ID
    #                       |      ________ Device Type
    #                       |     |     ___ Group ID
    #                       |     |    |
    #                       v     v    v
    command_topic: milight/0xD5F/rgb_cct/1
    state_topic: milight/states/0xD5F/rgb_cct/1
    platform: mqtt
    schema: json
    color_temp: true
    rgb: true
    brightness: true

  - name: "Living Room Ceiling"
    command_topic: milight/0xD4E/rgbw/2
    state_topic: milight/states/0xD4E/rgbw/2
    platform: mqtt
    schema: json
    color_temp: true
    rgb: true
    brightness: true

  - name: "BedRoom"
    command_topic: milight/0xD4E/rgbw/3
    state_topic: milight/states/0xD4E/rgbw/3
    platform: mqtt
    schema: json
    color_temp: true
    rgb: true
    brightness: true
```

### What is the model number of the device you're trying to control?

2 FUT019 bulbs, 3 RGB CW Bulbs(I think they're FUT012)

### What firmware version(s) have you tried? 
1.8.7

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
NodeMCU

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGBW, RGB+CCT

### Have you tried controlling the device with a physical remote?
Yes, physical remote works(and the hub can pick up the packets it sends). Model is FUT089, works with the FUT019 bulbs.  Info from sniffing:
````raw
fut089 packet received (9 bytes):
Raw packet: BB 34 B3 9C EF CE F4 04 06 

Decoded:
Key      : BB
b1       : 25
ID       : 5F8C
Command  : 01
Argument : 00
Sequence : 4F
Group    : 00
Checksum : 5F
````

 

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
{
  "version": "1.8.7",
  "variant": "nodemcuv2",
  "free_heap": 25920,
  "arduino_version": "2_4_2",
  "reset_reason": "External System"
}
```

#### /settings

```json
{
  "admin_username": "",
  "admin_password": "",
  "ce_pin": 4,
  "csn_pin": 15,
  "reset_pin": 0,
  "led_pin": -2,
  "radio_interface_type": "nRF24",
  "packet_repeats": 10,
  "http_repeat_factor": 1,
  "auto_restart_period": 1440,
  "mqtt_server": "kqmt-phoenixnas:1883",
  "mqtt_username": "",
  "mqtt_password": "",
  "mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
  "mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
  "mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
  "discovery_port": 48899,
  "listen_repeats": 3,
  "state_flush_interval": 10000,
  "mqtt_state_rate_limit": 100,
  "packet_repeat_throttle_sensitivity": 0,
  "packet_repeat_throttle_threshold": 200,
  "packet_repeat_minimum": 5,
  "enable_automatic_mode_switching": true,
  "led_mode_wifi_config": "Fast toggle",
  "led_mode_wifi_failed": "On",
  "led_mode_operating": "Fast blip",
  "led_mode_packet": "Flicker",
  "led_mode_packet_count": 5,
  "device_ids": [
    3423
  ],
  "gateway_configs": [
    [
      3406,
      12345,
      5
    ],
    [
      0,
      54321,
      5
    ]
  ],
  "group_state_fields": [
    "state",
    "brightness",
    "color_temp",
    "computed_color"
  ]
}
```

MunkeyBalls, 2020-01-21:
Can you pair the bulbs again by following the steps described [here](https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs)?

sidoh, 2020-01-26:
Hello,

Your hypothesis makes sense to me, although your guess is as good as mine as to why that happened.  Bulbs de-pair when they get an "on" packet from their remote within 5s of turning on.  They usually turn color to red to indicate they've been depaired.

Trying to repair makes sense to me.

Could also be an issue with the nRF board.  Maybe double-check that connections haven't come loose or corroded.