sidoh/esp8266_milight_hub #590: Hass.IO no Auto discovery or apperance in devices

quadragin, 2020-01-20:
I have started from scratch rebuilt my HASSOS.
I have Mosquitto Broker installed and running with a user name/password and access control list enabling unrestricted access to all topics.

The Milights are not auto added in the HA/Overview screen, nor does it show up in Configuration/Devices.

In a terminal session I run mosquitto_sub -h 192.168.107.30 -u .....-P .....-t 'milight/#' -v
I get this back in return when I issue a command in the espMH web interface
milight/states/0xE046/rgbw/2 ["state: "ON", "brightness": 20, "bulb_mode...etc..

**In the espMH**
MQTT topic pattern - milight/:device_id/:device_type/:group_id
MQTT update topic pattern - nothing
MQTT state topic pattern - milight/states/:device_id/:device_type/:group_id

**and have added this to my configuration.yaml**
mqtt:
broker: 192.168.107.30
discovery: true
discovery_prefix: homeassistant

light:
  - platform: mqtt
     schema: json
     name: "Deck_Stairs"
     state_topic: "milight/state/0xE046/rgbw/2"
     command_topic: "milight/command/0xE046/rgbw/2"
     brightness: true
     color_temp: true
     rgb: true
     effect: true
     effect_list:
        white_mode
        night_mode
        0
        1
        2
        3
        4
        5
        6
        7
        8

**My Mosquitto log looks like this:**
[10:57:27] INFO: Setup mosquitto configuration
[10:57:27] INFO: Found local users inside config
[10:57:27] INFO: Initialize Hass.io Add-on services
[10:57:27] INFO: Initialize Home Assistant discovery
[10:57:28] INFO: Start Mosquitto daemon
1579543048: mosquitto version 1.6.3 starting
1579543048: Config loaded from /etc/mosquitto.conf.
1579543048: Loading plugin: /usr/share/mosquitto/auth-plug.so
1579543048: ├── Username/password checking enabled.
1579543048: ├── TLS-PSK checking enabled.
1579543048: └── Extended authentication not enabled.
1579543048: |-- *** auth-plug: startup
1579543048: Opening ipv4 listen socket on port 1883.
1579543048: Opening ipv6 listen socket on port 1883.
1579543048: Opening websockets listen socket on port 1884.
1579543048: Opening ipv4 listen socket on port 8883.
1579543048: Opening ipv6 listen socket on port 8883.
1579543048: Opening websockets listen socket on port 8884.
1579543048: Warning: Mosquitto should not be run as root/administrator.
1579543049: New connection from 192.168.107.20 on port 1883.
[INFO] found mqtt on local database
1579543050: New client connected from 192.168.107.20 as milight-hub-1670881 (p2, c1, k15, u'mqtt').
1579543088: New connection from 192.168.107.30 on port 1883.
1579543088: Socket error on client , disconnecting.
1579543704: New connection from 172.30.32.1 on port 1883.
[INFO] found mqtt on local database
1579543705: New client connected from 172.30.32.1 as mosq-P7wey5hTiNNWZ390EI (p2, c1, k60, u'mqtt').

quadragin, 2020-01-20:
I did notice this in my nome-assistant.log
2020-01-20 12:57:50 ERROR (Thread-2) [homeassistant.components.mqtt] Unable to connect to the MQTT broker: Connection Refused: not authorised.
2020-01-20 12:57:51 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved

quadragin, 2020-01-20:
I removed the username and password from the mosquitto config, deleted the Access control list stuff and removed the mqtt user and password stuff from the setting in the web UI
my light now shows up on the overview screen but with no control
Mosquitto Log
1579552266: New connection from 192.168.107.20 on port 1883.
1579552266: New client connected from 192.168.107.20 as milight-hub-1670881 (p2, c1, k15).

quadragin, 2020-01-20:
Here is my configuration.yaml
light:
  - platform: mqtt
    schema: json
    name: "Deck_Stairs"
    state_topic: "milight/state/0xE046/rgbw/2"
    command_topic: "milight/command/0xE046/rgbw/2"
    brightness: true
    color_temp: true
    rgb: true
    effect: true
    effect_list:
      - white_mode
      - night_mode
      - 0
      - 1
      - 2
      - 3
      - 4
      - 5
      - 6
      - 7
      - 8
[image]

garmck, 2020-01-20:
From what you have presented, your configs do not match.
**espmh:**
> MQTT topic pattern - milight/:device_id/:device_type/:group_id
>MQTT state topic pattern - milight/states/:device_id/:device_type/:group_id

**Hass Config:**
>state_topic: "milight/state/0xE046/rgbw/2"
>command_topic: "milight/command/0xE046/rgbw/2"

You have added an S at the end of states and you are missing the /command/

Change your settings in the espMH to:
MQTT topic pattern
`milight/command/:device_id/:device_type/:group_id`
MQTT state topic pattern
`milight/state/:device_id/:device_type/:group_id`