sidoh/esp8266_milight_hub #591: Just getting Device ID 0000

sascha777, 2020-01-24:
Hi guys,

I just started with a milight hub to control two CCT led panels that run on 2.4 GHz and that I actually set via a remote control. The panels are bought from an internet store (lampenwelt), so it's definitivly not a Mi Light but I would like to give it a try to check if it's running on the mi protocol.

Problems:
- Sniffed traffic comes in very slow (with latency) and just a very few messages (didn't have a capacitor on hand yet)
- All messages report ID 0000
`rgb_cct packet received (9 bytes): Raw packet: 00 DB E1 23 66 CD C2 65 B1 
Decoded:
Key      : 00
b1       : 20
ID       : 0000
Command  : 01
Argument : 05
Sequence : 08
Group    : 00
Checksum : E6`
- In MQTT I do see some 
`{"status":"disconnected_unclean","firmware":"milight-hub","version":"1.10.5","ip_address":"xxx.xxx.xxx.xxx","reset_reason":"External System"}` between other normal messages

I run a precompiled esp8266_milight_hub_nodemcuv2-1.10.5.bin (did try d1_mini and esp12 as well) on a NodeMCU v3 (I read that this should be compatibe with v2) with a NRF24L01+PA+LNA SMA bought [here](https://www.amazon.de/gp/product/B00LNVJLZA/ref=ppx_yo_dt_b_asin_title_o01_s02?ie=UTF8&psc=1). 20cm Dupond wirering was done carefully and validated against the pinout layout sent with the NodeMCU.

about shows
`{"firmware":"milight-hub","version":"1.10.5","ip_address":"xxx.xxx.xxx.xxx","reset_reason":"External System","variant":"nodemcuv2","free_heap":16008,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}`

with setting
`{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"openhab.fritz.box","mqtt_username":"milighthub","mqtt_password":"","mqtt_topic_pattern":"milighthub","mqtt_update_topic_pattern":"milighthub/update","mqtt_state_topic_pattern":"milighthub/state","mqtt_client_status_topic":"milighthub/client","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":1000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"MID","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color"],"group_id_aliases":{}}`
tried with rf24_listen_channel with high, mid and low.

Having read the troubleshooting guide I am unshure if I have a bad radio board, my light doesn't run on the mi protocol or I did something else wrong.

Do you have any clue what this could be?
Thanks for helping

sidoh, 2020-01-26:
reset reason being `External System` seems to indicate the device is rebooting itself for some reason.  Serial logs might reveal more.

My best guesses are:

* Power supply issues (try with a known good supply).  The nRF board you're using draws more power than the ones with PCB antennae.
* Bad radio board

sascha777, 2020-01-28:
I tried to externally power the nRF but without success. 
Will order a new board and maybe switch to an LT8920. When the components arrived I will report back.

sidoh, 2020-01-30:
I'd check serial logs too.

Wouldn't recommend using LT8900.  Listening features aren't as robust for it.

sascha777, 2020-02-01:
Meanwhile I tried another NRF24L01 with PCB antenna but still getting no results. 
Will check the serial logs, but my feeling is that the light aren‘t mi compatible.

sascha777, 2020-02-05:
I set up a second device and both devices are now communicating fine.
Unfortunately neither my first led ceiling light nor another led panel seem to work. Both paired remote controls can't be sniffed :-(

Is there another thing that I could try to get them running?

sidoh, 2020-02-09:
Ah, if the factory remotes you've got aren't being detected, it's probably not compatible unfortunately.

sascha777, 2020-02-10:
Ok, thanks for your help. Maybe I will order some real mi lights in the near future. 
Keep the project going!

closed 2020-02-10 by sascha777