sidoh/esp8266_milight_hub #592: Mqtt Settings where to edit ?

gcharles81, 2020-01-26:
Hello Thanks very much for this great project , I have nodemcu setup and working fine with web UI

I would like to use the mqtt feature since I already have mosquitto running on my home automation , but I got confused where to edit the MQTT settings in the source code 

Should I edit the settings in the Settings.cpp file as I do not find them  as per below from the readme ? thanks in advance
 
## MQTT

To configure your ESP to integrate with MQTT, fill out the following settings:

1. `mqtt_server`- IP or hostname should work. Specify a port with standard syntax (e.g., "mymqttbroker.com:1884").
1. `mqtt_topic_pattern` - you can control arbitrary configurations of device ID, device type, and group ID with this. A good default choice is something like `milight/:device_id/:device_type/:group_id`. More detail is provided below.
1. (optionally) `mqtt_username`
1. (optionally) `mqtt_password`

poudenes, 2020-01-26:
[image]

You edit it here... just in the webui

sidoh, 2020-01-26:
Thanks, @poudenes!

Yep, all settings are configurable in the web UI.  If you get stuck again, the README goes into a fair amount of detail re: setup.

closed 2020-01-26 by sidoh