sidoh/esp8266_milight_hub #597: RGBW light with "B1"=18

normen, 2020-02-13:
I don't seem to be able to control lamps, I can receive signals from the remote/iBox but when sending signals from milight-hub the lamps don't react.

I noticed that "B1" in the sniffing log is different when I control the lamp through my iBox and when I try control it from milight-hub.

What the lamp sends:
````
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 69A1
b1            : 18
b2            : 02
b3            : 06
Sequence Num. : 0C
````
What milight-hub sends:
````
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 69A1
b1            : 00
b2            : 02
b3            : 06
Sequence Num. : 0B
````

### What is the model number of the device you're trying to control?
MiLight RGBW Series E27 bulb (not in catalog?)

### What firmware version(s) have you tried? 
1.10.5

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
d1_mini

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGBW

### Have you tried controlling the device with a physical remote?
No, with an iBox2

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
{"firmware":"milight-hub","version":"1.10.5","ip_address":"192.168.2.190","reset_reason":"External System","variant":"d1_mini","free_heap":17728,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### /settings

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[12891,27041],"gateway_configs":[],"group_state_fields":["state","brightness","computed_color","mode","color_temp","bulb_mode"],"group_id_aliases":{"Fenster":["rgbw",27041,2],"Kinderzimmer 1":["rgbw",12891,2]}}
```

normen, 2020-02-13:
I should mention that I CAN control a RGBW controller box that has a LED Strip connected just fine so my RF module is working.

normen, 2020-02-13:
I tried changing this line:
https://github.com/sidoh/esp8266_milight_hub/blob/50e91bf63d6249348300a9c7f502f2ced949a55a/lib/MiLight/RgbwPacketFormatter.cpp#L19
to `0x18` and the commands look similar now in the log but the bulbs still won't work.

normen, 2020-02-13:
In case its a hardware issue, I ordered a new transceiver module, I'll report back when it arrives.

sidoh, 2020-02-14:
Hiya @normen,

As you point out, if the radio is working with other devices, it's unlikely to be the culprit.

You can try using the [raw packet REST routes](https://sidoh.github.io/esp8266_milight_hub/branches/latest/#tag/Raw-Packet-Handling) to send a clone of a packet your remote sends.  Note that you'd need to reset the power on your bulb as it'll ignore packets with the same packet ID as the last one it received.

If this doesn't work, the problem is deeper and more mysterious.  My guess if that were to be the case is there's stuff outside of the milight payload that your bulbs are sensitive to.  Basically impossible for me to debug that without access to the physical devices (and -- honestly -- time I don't expect to have any time soon).

normen, 2020-02-14:
Thanks for the answer, sadly no dice either. 🤷‍♂

A new transceiver module should arrive on monday, lets see.. In [another issue](https://github.com/sidoh/esp8266_milight_hub/issues/498) somebody seemed to have a similar problem with FUT104 bulbs only and it was indeed a faulty transceiver board. I do seem to notice issues with my WiFi when the transceiver is powered on, so that might be a hint. I'll report back.

..and btw cheers for this project - it's awesome!

normen, 2020-02-15:
I got a new radio module, it works. WTF

The old and new one look EXACTLY the same: https://www.amazon.de/gp/product/B07P95X6HM/

I don't know where I sourced the old one, it was in a bulk buy of electronics parts I once did.

So again to reiterate for others who might have the same issues: I was able to control my RGB(W) strips (FUT037/038) and receive all messages but my bulbs would not react (FUT104). The only other hint was that my WiFi was bogged down a bit when the faulty module was powered.

wow. dem clonez o_O

richms, 2020-02-15:
How close was that module to the ESP? I have found that those ones with the power amplifier on them are real sensitive to other boards near them. Bluetooth or wifi and it really messes them up. I put one on a pi hat for another project and nothing worked properly, weird because the non power amp one worked fine and it worked fine in breadboarding. Finally extended the wires 200mm away from the wifi on the pi and it worked fine. Move it back above the wifi on the pi and erraticness resumed.

normen, 2020-02-15:
Not too close, this isn't the same board but basically the same distance: https://i.imgur.com/ZDkKg8P.png

...and the current one is at at an even shorter distance in a small box with the ESP, working far better with a wider coverage than the stupid iBoxes :)

Thanks again for this project @sidoh!