sidoh/esp8266_milight_hub #598: GTOP support

ligius-, 2020-02-24:
There are some LED-panels from GTOP that have temperature control (CCT) and work with a PL1167. One of my 4 remotes has a 2-chip solution, so I was able to sniff the SPI registers for the communication, but I'm having several questions:
- how are the preamble and syncwords configured?
- what are the channel numbers?

As a quick protocol overview, it has 10 data bytes:
1-2 - remote id?
3 - key pressed, usually the higher nibble is an 8
4-6 - seem to be fixed
7 - brightness
8 - temperature
9-10 - some kind of toggling or counter, does not make much sense, not a CRC

Relevant PL1167 registers:
20 58 00	data conf len_preamble, len_sync 64b, len_trailer 4b, nrz law, no fec
24 03 80	syncword 0 [15:0]
25 03 80	syncword 1 [31:16]
26 5A 5A	syncword 2 [47:32]
27 03 80	syncword 3 [63:48]
29 BC 00	misc ‭1011110000000000‬ crc on, scramble off, en_pack_len on

Transmitting:
23 03 00 - scramble seed 0, max retransmit 3?
07 01 14 - transmit en, ‭ch = 0010100 (2402 + 20?)‬

I don't have access to a 2.4GHz analyzer and I somehow cannot get the nRF24L to sniff the packets, perhaps I need a different channel selection. If the nRF starts from 2400 and the PL1167 from 2402, I assume CH20 on the PL would be 22 on nRF?
The PL datasheet says that "LSB bits of sync word are first", not sure what that means w.r.t to MiLight configuration.
MiLight allows to configure syncwords 0 and 3, but, on my unit, syncword 2 is actually different.

I can provide more detail, but thought I would open a ticket while I playing around trying to get it working.

ligius-, 2020-02-24:
I took a look at the implementation and the original article, I think I understand how it should be configured.
The initial implementation uses syncword0 and 3 because the syncword is set to 32 bits. For 64 bits, syncword0 and syncword1 can be used (in my case 0x0380 0x0380) and it should be able to match the packets. The preamble stays the same at 3 bytes. It would mismatch on the packet length field but it should still capture the packets.
However, even with those changes I could not get it to capture anything from my remotes.

ligius-, 2020-02-24:
Well, it took me a while to figure out that the channels listed are for transmitting only, for receiving only the first one is used.
Got it receiving signals with this config:
```
  MiLightRadioConfig(0x0380, 0x0380, 15, 20, 20, 20, 0xAA, 0x05),  // GTOP
```
SYNCWORD_LENGTH was set to 4 since the code is not ready to handle 64-bit syncwords.

```
Packet received (19 bytes): 5A 5A 80 03 A5 F0 AF 58 4B 10 00 28 C3 28 0D A0 79 FF 17 
Failed CRC: expected 5721, got 17FF
```
The first 4 bytes are still syncwords. By running the original program (openmilibas016.ino) the data gets a bit further:
```
Packet received: 035A5A8003A5F0AF584B100028C3080D
Packet transformed: A50538500AFF8AB5040180328CD0
```
The actual payload, without the end CRC is: `0AFF8AB504018032####` which is achieved by RSH(4).

I can avoid the shifting with a hack that allows SYNCWORD_LENGTH to be 5 by setting the trailer to be the lower byte of syncword2, plus another hack that shifts the input buffer by 4 bytes:
```
  MiLightRadioConfig(0x0380, 0x0380, 13, 20, 20, 20, 0xAA, 0x5A),  // GTOP
...
  for (int inp = 0; inp < _receive_length - 4; inp++) {
    tmp[outp++] = reverseBits(tmp[inp + 4]);
  }
...
Packet received (12 bytes): 0A FF 8A B5 04 01 80 32 8C D0 00 2A 
Failed CRC: expected 36E2, got 2A00
```

The CRC check still fails, I haven't investigated why.

ligius-, 2020-02-27:
Final comment, before I move to a different project.

I did all the modifications so the packets are read and written successfully. There's a high CRC error rate when decoding, perhaps 80%.

A raw and a decoded packet:
Packet received (0 bytes) RAW: A5 A0 1C 0A 50 FF 51 AD 20 80 01 4C 55 87 00 4D 7D 
Packet received (13 bytes):   A5 05 38 50 0A FF 8A B5 04 01 80 32 AA E1 00 B2 BE 

From the decoded packet, the first 4 bytes have to be dropped since they are part of the syncword. 0x0A is the length, up to 0x00. The last two bytes are the CRC, works with the same polynomial.

Sending exactly the same message back through the nRF, through the same pipe, does not seem to work.

sidoh, 2020-02-27:
Hi @ligius-, apologies for the lack of response.  Thanks for digging into this, and impressed with how far you got!

I unfortunately don't have time to dig into a new protocol right now.  

There have been a few really weird cases where people have had nRFs that can receive data, transmit data that another nRF can see, but that milight hardware doesn't respond to (see #498 for a detailed debugging log).  Replacing the module fixed their problems.  My knowledge of the stack beneath the nRF layer is pretty much zilch, so no idea what's going on (would love to learn, just don't have a ton of time these days).  Perhaps worth a shot here.

Another idea -- set up another nRF to make sure the packets you're sending from the first are showing up.

closed 2020-11-01 by sidoh

kiddyfurby, 2021-04-12:
@ligius- any chance you could share a patch file of your work so that I can learn and possibly continue GTOP support?

ligius-, 2021-04-12:
Hi, not sure what the best way is to share this. I have a file with the SPI commands that I'm sending directly to the chip and gave up for now on using the nRF24.

[ledpanel.yaml.txt](https://github.com/sidoh/esp8266_milight_hub/files/6298826/ledpanel.yaml.txt)

[LT8900Light..h.txt](https://github.com/sidoh/esp8266_milight_hub/files/6298815/LT8900Light.h.txt)

I'm attaching the milight patch file but it's a mess and have no idea anymore what needs to be there and what doesn't.
[patch.txt](https://github.com/sidoh/esp8266_milight_hub/files/6298844/patch.txt)

kiddyfurby, 2021-04-14:
Thank you so much - that's way more help than I expected!
lib/MiLight/GtopPacketFormatter.h is missing so the patched project is not buildable.
I am going to get your esphome port running as a first step.

ligius-, 2021-04-14:
Sorry, guess I forgot to add it to my repo. Here it is, but again, the entire branch is broken as I played around a lot with it. Definitely not even close to production-ready:

[GtopPacketFormatter.h.txt](https://github.com/sidoh/esp8266_milight_hub/files/6312690/GtopPacketFormatter.h.txt)