sidoh/esp8266_milight_hub #599: Feature request: External button support

wormiedk, 2020-02-24:
It would be great to be able to add push buttons that would either
- launch a specific scene
- generate an mqtt message (e.g doorbell)

sidoh, 2020-02-28:
Hi @wormiedk, request makes sense and is reasonable.  That being said, does seem like it's the sort of thing that's susceptible to feature creep.

What does "specific scene" here mean?

mozzhead164, 2020-09-19:
@sidoh I have a use case for this. I have a cool white strip across the top of the garage controlled using a single colour Mi-Light controller.

This comes on with a time clock so is on between dusk and dawn. I would like for it to have a high-low feature. Low brightness under usual circumstances. When the garage door is triggered, I would like to use a voltage free contact to trigger the ESPMLH to turn the brightness up to full, then after a minute or so turn back down to the lower brightness. 

Is the only current way to do this via MQTT?