sidoh/esp8266_milight_hub #601: All settings lost

idfxken, 2020-02-26:
### Describe the bug

When I came home, my lights didn't function, and after checking the milighthub has lost all settings
(settings.json contains only "{ }" )
### Steps to reproduce

Leave it running long enough.

### Setup information

ha with autodiscovery , mixed cct, cct+rgbww, + rgb lights (around 7 groups)

#### Firmware version
1.10.5

#### Output of http://milight-hub.local/about

```

firmware | "milight-hub"
-- | --
version | "1.10.5"
ip_address | "192.168.1.24"
reset_reason | "External System"
variant | "nodemcuv2"
free_heap | 19200
arduino_version | "2_4_2"
queue_stats |  
length | 0
dropped_packets | 0

```

#### Output of http://milight-hub.local/settings

```admin_username | ""
-- | --
admin_password | ""
ce_pin | 4
csn_pin | 15
reset_pin | 0
led_pin | -2
radio_interface_type | "nRF24"
packet_repeats | 50
http_repeat_factor | 1
auto_restart_period | 0
mqtt_server | ""
mqtt_username | ""
mqtt_password | ""
mqtt_topic_pattern | ""
mqtt_update_topic_pattern | ""
mqtt_state_topic_pattern | ""
mqtt_client_status_topic | ""
simple_mqtt_client_status | false
discovery_port | 48899
listen_repeats | 3
state_flush_interval | 10000
mqtt_state_rate_limit | 500
packet_repeat_throttle_sensitivity | 0
packet_repeat_throttle_threshold | 200
packet_repeat_minimum | 3
enable_automatic_mode_switching | false
led_mode_wifi_config | "Fast toggle"
led_mode_wifi_failed | "On"
led_mode_operating | "Slow blip"
led_mode_packet | "Flicker"
led_mode_packet_count | 3
hostname | "milight-hub"
rf24_power_level | "MAX"
rf24_listen_channel | "LOW"
wifi_static_ip | ""
wifi_static_ip_gateway | ""
wifi_static_ip_netmask | ""
packet_repeats_per_loop | 10
home_assistant_discovery_prefix | ""
wifi_mode | "n"
default_transition_period | 500
rf24_channels |  
0 | "LOW"
1 | "MID"
2 | "HIGH"
device_ids | []
gateway_configs | []
group_state_fields |  
0 | "state"
1 | "brightness"
2 | "computed_color"
3 | "mode"
4 | "color_temp"
5 | "bulb_mode"
group_id_aliases | {}
```

### Additional context
Stupid me forgot to backup the settings aswell...

sidoh, 2020-02-26:
Thanks for reporting.  This has come up a couple of times (#600, probably #523).

I have unfortunately not found a reliable way to reproduce the issue, which obviously makes it tricky to fix.

My best guess so far is that there's a memory leak, and something in the chain of settings parsing/loading, serializing/saving is barfing and not handling the failure gracefully.

Will definitely look into this when I have the chance, but if anyone has a reliable way to reproduce the issue, that'd make it 100x easier to fix.

poudenes, 2020-02-27:
Ill comment in #600 what I think it can be in my case

csigabit, 2020-03-09:
I have the same issue on multiple locations.
To reproduce:
-Leave it run loooong enough
-Change some setting or upload a settings.json
-Soft reboot
Settings and Devices are gone, but it will stay connected. Hw reset button or soft reboot doesn't help, I have to power cycle the unit and restore settings to make it work again.

idfxken, 2020-03-20:
It just happened for the 3rd time: 
lighthub/about:

```
{"firmware":"milight-hub","version":"1.10.5","ip_address":"192.168.1.24","reset_reason":"External System","variant":"nodemcuv2","free_heap":6408,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":47}} 
```

No idea if this is usefull:
While all my settings are completely gone,
all the lights can still be controlled perfectly except 1 group, 
the exact same group alerted me the last time aswell.

sidoh, 2020-07-01:
I'm going to close this as a duplicate of #600.

closed 2020-11-01 by sidoh