sidoh/esp8266_milight_hub #603: some things can't be done in one command

csigabit, 2020-03-09:
### Describe the bug

Maybe it's just my bulbs, but
- set_white with tempereture setting doesn't work in one command
- if a bulb is off, then turning it on and changing colormode doesn't always work in one command
- changing colormode and transition in one command makes the transition flicker and not end up in desired color or temp

### Steps to reproduce

- turn on the bulb and switch to a color
- {"command":"set_white","kelvin":5,"brightness":80,"transition":2}
--> The temperature flickers and won't end up in the desired cool

NEXT

- turn on the bulb and switch to a color
- turn off the bulb
- {"status":"ON","command":"set_white","kelvin":5,"brightness":80,"transition":2}
--> Sometimes it switches back to color mode quickly, the temperature flickers and won't end up in the desired cool

### Expected behavior

Do at once, because right now what I do to make it work and eye friendly:
- Store states, because we need to know the status and colormode
- If the **bulb is off** then ( {"status":"ON","brightness":1}     +     **600ms delay**)
after that
- If there's **change in colormode** then ( {"command":"set_white","brightness":1}  or  {"hue":{{hue}}}     +     **600ms delay**     +     remove transition only from next command, because it flickers with colormode change)
after that
- The bulb is on and it's colormode is same as the command, so I can change temp or color ({"kelvin":{{tem}},"brightness":{{bri}},"transition":2}  or  {"hue":{{hue}},"brightness":{{bri}},"saturation":"{{sat}}","transition":2.5})

I don't want to resend commands and I want every command to end up with desired light settings, so I'm not reducing this delay any more or changing this method until this is fixed. Finding the solution or integrating the necessary delays to the source would be nice, because theese bulbs just won't do things in one command for me.

### Setup information

#### Firmware version
1.10.5 (d1_mini)
#### Output of http://milight-hub.local/about

```json
"{"firmware":"milight-hub","version":"1.10.5","ip_address":"192.168.21.6","reset_reason":"External System","variant":"d1_mini","free_heap":14480,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":106}}"
```

#### Output of http://milight-hub.local/settings

```json
"{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":24,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.21.3:1883","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"milight/:device_alias/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_alias/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:device_alias/:device_type/:group_id","mqtt_client_status_topic":"milight/LWT","simple_mqtt_client_status":false,"discovery_port":0,"listen_repeats":0,"state_flush_interval":600000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":400,"packet_repeat_minimum":16,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Off","led_mode_packet_count":0,"hostname":"milight-hub-6","rf24_power_level":"MAX","rf24_listen_channel":"HIGH","wifi_static_ip":"192.168.21.6","wifi_static_ip_gateway":"192.168.21.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":4,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":160,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[4478,10227,10343,12893,16151,17972,18649,21308,22853,27173,29723,34991,38250,44115,46648,52530,55080,60945],"gateway_configs":[[60945,8899,6]],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type","hex_color"],"group_id_aliases":{"Bathroom":["rgb_cct",55080,1],"Bedroom":["rgb_cct",38250,1],"Diner":["rgb_cct",4478,1],"Entry":["rgb_cct",10343,1],"Garden":["rgb_cct",22853,1],"Guest":["rgb_cct",16151,1],"HallBack":["rgb_cct",27173,1],"HallFront":["rgb_cct",18649,1],"Kitchen":["rgb_cct",12893,1],"Laundry":["rgb_cct",29723,1],"Livingroom":["rgb_cct",34991,1],"Office":["rgb_cct",46648,1],"Rack":["rgb_cct",21308,1],"Stairs":["rgb_cct",44115,1],"Storage":["rgb_cct",10227,1],"UpstairsBack":["rgb_cct",52530,1],"UpstairsFront":["rgb_cct",60945,1]}}"
```

### Additional context

sidoh, 2020-03-09:
Try with just `{"kelvin":5}` (without `{"command":"set_white"}`).  For most bulbs, the set_white command is just forcing kelvin to some value (sometimes the last known one).

csigabit, 2020-03-09:
Thanks for the quick solution! That was it. No metter what, theese work:
{"status":"ON","hue":50,"brightness":50,"saturation":"50","transition":2}
{"status":"ON","kelvin":50,"level":50,"transition":2}
I re updated my nodered flows and the domoticz plugin also so it can change kelvin now. My fork:
https://github.com/csigabit/espmilighthub-domoticz

closed 2020-03-09 by csigabit