sidoh/esp8266_milight_hub #605: FUT095 Remote and my bulbs are FUT018 but nothing happens

NoseBagUK, 2020-03-12:
Hi,

I'm having problems getting mine working and hoped you could help.

I've read through the Troubleshooting guide and it hasn't helped.

I bought one of the excellent pre-made hubs and I'm pretty sure the hardware is fine (when I built my own I got a similar result.)

My remote is a FUT095 and my bulbs are FUT018, these don't appear to be on your supported bulb page, but I see from this issue that the remote should be supported...
https://github.com/sidoh/esp8266_milight_hub/issues/495

In the image below the top packet is what my remote sends when I click the “All On” button and the bottom packet is what was sniffed when I sent when I clicked the “All On” button in the web app.
[image]
So it looks like it isn't sending the right commands. 

### What firmware version(s) have you tried? 
1.10.5

### Which ESP8266 board are you using? (nodemcu, d1_mini, etc.) 
nodemcuv2

### Which radio type are you using? (RGBW, RGB+CCT, etc.) 
RGBW

### Have you tried controlling the device with a physical remote?
Yes, works fine, FUT095 

### Output of http://milight-hub.local/about and http://milight-hub.local/settings 

#### /about

```json
{"firmware":"milight-hub","version":"1.10.5","ip_address":"192.168.50.15","reset_reason":"Power on","variant":"nodemcuv2","free_heap":17792,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

#### `/settings`

```json
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[2249],"gateway_configs":[],"group_state_fields":["state","brightness","computed_color","mode","color_temp","bulb_mode"],"group_id_aliases":{"office":["rgbw",2249,1]}}
```

NoseBagUK, 2020-04-14:
I've been thinking about this again lately and figured I can just implement the codes that it appears I need for my remote. Can anyone point me in the right direction for the place in the code I need to do this? I'm a programmer but not a C person so I could do with a pointer (no pun intended).

sidoh, 2020-04-15:
Hiya @NoseBagUK, and apologies for the delayed response (and appreciate the bump/offer to help).  Have had a hard time keeping up with issues.

(lol @ pointer)

If I'm following, the issue is:

* Your remote can successfully control your bulb
* EspMH is picking up your remote
* EspMH appears to be sending packets
* But the bulb does not respond to packets sent by EspMH

is that correct?

it's worth verifying that packets are *actually* getting sent if you haven't already.  Two ways to do this:

* set up a 2nd espMH install and listen for packets from the other.
* use the [`/raw_commands` REST endpoint](https://sidoh.github.io/esp8266_milight_hub/branches/latest/#tag/Raw-Packet-Handling/paths/~1raw_commands~1{remote-type}/post) to send a copy of a packet from your remote.  Packets have IDs, and bulbs ignore packets with IDs near the last one it received.  I'd recommend doing this:
  * Turn sniffing on on espMH
  * Turn bulb off with remote, copy packet from espMH
  * Turn bulb back on by toggling physical power
  * Send raw packet via /raw_commands endpoint -- bulb should turn off (if it doesn't, there's a hardware issue)

The relevant packet formatter for this protocol is here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/RgbwPacketFormatter.cpp

Tontze, 2020-04-22:
Im hijacking this a little bit, as i see behaviour i have wondered on the picture attached on first post. Im receiving decimal ID from remote, but espMH sends hex out to bulp. Is this just normal or do you need to convert something in some point ?