sidoh/esp8266_milight_hub #606: Home Assistant Groups vs Hub Groups

zestyphresh, 2020-03-19:
I'm just curious if there is a performance benefit to doing light groups in the hub versus in Home Assistant. Being able to emulate multiple hubs makes it easy to link bulbs individually on one and as a single group in another (or simply using the [all] tag). In my mind this reduces incoming bandwidth as there will be less messages but does it have any performance benefit at the radio end?

PetricaM, 2020-03-26:
Although I think there would be a performance benefit from sending one single command for the entire room (vs. let's say 4 commands if having 4 individual lights), the real improvement would be in the aesthetics department as all lights would change instantly. Even with a two lights group set in HA you would still get a noticeable delay; add in more individual lights to the HA group and you'll need in excess of 2-3 seconds for all the lights to change.

My setup (at least for the bedroom and the living, as for the other rooms I don't see any real benefit) is the following:
- all lights in the room are paired to group 1 of a B0 remote (wall mounted, single group); in HA I have this group set for the lights in the room; the wall mounted remote is near the Zigbee smart switch I usually use; I use this B0 physical device only case there are problems with HA, EMH or wifi connecting the two (most of the time lights are controlled by PIR sensors or the Zigbee smart switches);
- each individual light is paired to a group from 1 to 4 of a physical FUT092 remote (handheld, 4 groups); this remote I only use in exceptional cases, or when kids (both small and grownups :) ) want scenes with different colors for each light; in 90% of the time all lights have the same colors (that is white or some variation of sandy brown depending on the time of the day; I have also set the circadian lighting in HA; after sunset also the brightness of the lights decreases); in order to decrease load on EMH the flow for the scenes with different colors for each light is as following:
      - send on command for B0 group; if lights are on, there's no change; if the lights are off, they would turn on with brightness set at 1 (I have a script that changes brightness to 1 before turning lights off in order to prevent turning at full brightness);
     - send color changes for each group from 1 to 4 of FUT092 remote; I could use group 0 from this FUT092 remote for the initial on command, however, there would be a lot of traffic on the state topic from EMH if using group 0;
- all lights in the house are paired to group 1 of a FUT088 remote (handheld, one group); this way I can walk in a room and, if the lights are on, then I can change the color in that room alone/all the rooms that have the lights on (if I press on off then the lights in that room would turn off; however, if I press the on key, then all the lights in the house would turn on);
- all lights in the house are paired, per room, to groups 1 to 4 of a B4 remote (wall mounted) near the entrance door; in the beginning I thought to use a B8 but it would get too complicated; the reason of using a physical native Milight remote instead of a tablet is to be able to add redundancy case any link between HA -> Router -> EMH would fail.

tbrasser, 2020-05-30:
hmm, but then how do you get the state (colours etc) synced to HA? (@previous comment)

PetricaM, 2020-05-30:
If you're referring to sending commands to a group set in ESPHM and then syncing the group's state to individual lights states in HA, then it is easy done in HA by forwarding the payloads from the group's **update** topic to individual lights **command** topics.

closed 2020-11-01 by sidoh