sidoh/esp8266_milight_hub #607: Home Assistant and MQTT device renaming

burton666, 2020-03-22:
I admit I am new to both mqtt and home assistant. But I managed to add some zigbee devices using a zigbee2mqtt device to home assistant and it works great. I can also rename all devices in home assistant.
I have also managed to add my milight devices to home assistant using the milight_hub and mqtt. But I can't rename any devices in HA because it says they do not have unique ids or something like that. So the only way for me to rename my devices is to unpair them and then pair again using the correct name. Is there a way to add the milight devices and also be able to rename them in HA ?

PetricaM, 2020-03-26:
You can manually add unique IDs in the light setup file

https://www.home-assistant.io/integrations/light.mqtt/#unique_id

You can use an UUID generator to get the unique ID.

burton666, 2020-03-26:
Ok, but I guess that I can not use auto discovery in that case?

PetricaM, 2020-03-26:
I don't know if it works with auto discovery (each integration should provide its own unique id) as I have everything set manually for MQTT. 
In order to avoid restarting HA after renaming a device I can either use customization (in yaml) or unique id (from gui).
Perhaps Chris can help.

sidoh, 2020-04-01:
Adding a unique ID in the discovery client is probably pretty easy:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MQTT/HomeAssistantDiscoveryClient.cpp#L42

Should be sufficient to make it a function of the device id, bulb type, and bulb id.