sidoh/esp8266_milight_hub #609: Auto discovery with multiple espMH devices

agoralive, 2020-03-30:
Hi

I have a setup where I have 5 espMH devices all listening on the same topics in order to get 100% reach around my house and eliminate dead spots.

I'm trying/considering Home Assistant (not the OS - but installed on my Rpi). I have everything setup and it's running perfectly through Node-Red and the Dashboard.

Do I need to setup device names/aliases for each remote, bulb and zone on ALL my espMH?

I ask because autodiscovery leaved me with a missing unique_id and is therefore not controllable. I just get the message that the bulb is unavailable

PetricaM, 2020-03-31:
You don't need to setup friendly names on all 5 hubs but only for a single EMH (if you would receive state updates from all 5 hubs you would only increase load on the MQTT broker and there is no real benefit as it would be the same payload received by all EMH allover again).

I think autodiscovery and unique_id need to be implemented in EMH code.

Alternatively, just set everything manually, add the unique id (you can use an uuid generator) and don't set the topic for availability as that one comes from the EMH, not from the bulb (if you have 5 gateways you would probably expect to have at least one of them picking up the payloads so the availability topic is not really needed).

sidoh, 2020-04-01:
ESPMH _should_ be using the chip ID to disambiguate the client -- https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MQTT/HomeAssistantDiscoveryClient.cpp#L131

With a setup like this, though, I'd personally probably steer clear of auto-discovery and just use YAML definitions.

If you are attached to auto-discovery, @PetricaM's suggestion makes sense.  Only set up auto-discovery on a single node, and leave it blank for the others.  Use the same MQTT command topic for everything, but only publish state from one of them.  (Keep in mind that even with this -- unless there's a lot of distance between nodes -- you'll probably get some extra noise from the master picking up traffic from the others)

agoralive, 2020-04-02:
Thank you both for your answers.... And you are right in most of the things you write.

I do not se much "noise" even though they all broadcasted their state, but those things a side, because I figured out how the Autodiscovery worked, and this also presented another issue which I corrected by using node-red....

The "Client Status Messages Mode" as your write needs to be set at Simple in order to use Auto Discovery, while this might be true, I have found that when the espMH restarts they sometimes sends out the disconnected message after the connected message, leaving homeassistant in a state where it thinks that the clients are unavailable.

In node-red what I did was mqtt-in -> change disconnected to connected -> mqtt-out

Import this into Node-Red, if tha's easier:

`[{"id":"33cfb1f.3a51c4e","type":"mqtt in","z":"99d896d7.deeb38","name":"","topic":"milight/client_status/#","qos":"2","datatype":"auto","broker":"f077ffa9.24771","x":460,"y":3120,"wires":[["6323d061.2c755"]]},{"id":"6323d061.2c755","type":"switch","z":"99d896d7.deeb38","name":"","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"disconnected","vt":"str"},{"t":"else"}],"checkall":"true","repair":false,"outputs":2,"x":630,"y":3120,"wires":[["17daf0f0.73fc0f"],[]]},{"id":"17daf0f0.73fc0f","type":"change","z":"99d896d7.deeb38","name":"","rules":[{"t":"change","p":"payload","pt":"msg","from":"disconnected","fromt":"str","to":"connected","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":820,"y":3120,"wires":[["44ce5e61.29bac"]]},{"id":"44ce5e61.29bac","type":"mqtt out","z":"99d896d7.deeb38","name":"","topic":"milight/client_status","qos":"2","retain":"true","broker":"f077ffa9.24771","x":1030,"y":3120,"wires":[]},{"id":"f077ffa9.24771","type":"mqtt-broker","z":"","name":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]`

I do not know if this can be setup and running while doing autodiscovery of new devices?????

Power is not always reliable here, so my router/internet sometimes disconnect and I have my espMH set to restart every 60 minutes, so I encountered this unavailable issue quite a lot!!!!!

With the above corrected I have now for 2 days not see unavailable in my HA lovelace

sidoh, 2020-04-11:
> I have found that when the espMH restarts they sometimes sends out the disconnected message after the connected message, leaving homeassistant in a state where it thinks that the clients are unavailable.

This definitely unexpected, and not something I've seen before.  It shouldn't be possible -- MQTT is supposed to guarantee message ordering, and I don't see how a disconnect could come after a connect without the real state of the MQTT client being disconnected.

Is it possible the multiple espMH instances are interfering with each other?

Would definitely recommend using static YAML configs over auto discovery in this case.

sidoh, 2020-11-01:
Closing as this is old, please feel free to re-open if it's still an issue.

closed 2020-11-01 by sidoh